KonradIT/goprowifihack #5: Streaming Delay

marksbren, 2015-03-22:
I recently tried streaming from a GoPro to my laptop and am disappointed with the video stream delay. Is there anyway to solve this?

g00ds0n, 2015-03-23:
Which GoPro do you have? My GoPro 3 had about a 5 second delay when streaming. It's better to use the HDMI port than it is to use the streaming feature of the camera.

marksbren, 2015-03-23:
I have the Hero4 and am using the commands in the Hero4.md file. I understand that an HDMI connection would be best, but would really like to fine a way to livestream over Wifi.

It is obviously doable since the GoPro iPhone App streams the camera in realtime and without the same delay. I was just curious if anyone had figured out the settings/configuration needed to eliminate the delay.

makk909, 2015-04-01:
I am trying to livestream Hero4 on android but not able to do so. Is there any special player required to livestream it on android. Also is there any way to livestream Hero4 on vlc in windows???

sravan953, 2016-03-12:
@makk909 I understand that the streaming is over UDP protocol, and that Android does not support this? Any updates?

silent3rror, 2016-04-21:
Try using ffplay:
`ffplay -fflags nobuffer -f:v mpegts -probesize 8192 udp://:8554`
[https://www.reddit.com/r/gopro/comments/2md8hm/how_to_livestream_from_a_gopro_hero4/cr1b193](url)

closed 2016-04-29 by KonradIT

videopro, 2016-09-29:
I tried but it's not working, did all this gopro live streaming hdmi - https://goo.gl/OpxjZi

KonradIT, 2016-09-29:
@videopro what camera? HERO4?

silent3rror, 2016-09-29:
@videopro Please paste the output from the command. As Konrad has said, what GoPro platform are you using, and what computer platform are you using (Windows, Mac, Linux)