KonradIT/goprowifihack #16: Secondary modes not working

digaus, 2015-11-24:
Somehow the secondary modes for the GoPro Hero 4 are not working... Also the Protune Settings are only working for the video mode and not for Photo or MultiShot :/ tried to find the settings but no succes

KonradIT, 2015-12-13:
what firmware?

digaus, 2015-12-17:
HD4.02.03.00.00

ztzhang, 2016-01-25:
It seems that the urls are incomplete for current firmware. I sniffed the GP Hero4 urls with the latest firmware, please refer to the latest issue.

depodefi, 2016-02-27:
Is there any progress on solving that issue?

ztzhang, 2016-02-27:
You can try this out: https://github.com/ztzhang/GoProWifiCommand

depodefi, 2016-02-28:
It's working, although you got some typos _submode_ instead of _sub_mode_

ztzhang, 2016-02-29:
It seems to be a firmware issue. Some firmware uses sub_mode instead.

KonradIT, 2016-04-23:
I updated the repo with the latest URLs from the latest FW (HERO4 Black 3.00 July 30, 2015)

closed 2016-04-23 by KonradIT