KonradIT/goprowifihack #21: 720p live streaming support (like Periscope)

Duffycola, 2016-03-08:
Do you have any idea how Periscope managed to achieve 720p live streaming? Here are some interesting comments: http://edbrophy.com/periscope-gopro-clearing-up-the-misinformation/

KonradIT, 2016-04-23:
I will have access to an iOS device next week so I will investigate it

KonradIT, 2016-05-02:
Looks like if you do a UDP wakeup call it starts the stream at 720p 

Tested: no

dvize, 2016-06-01:
Is there any more information on this?

sfarges, 2016-06-23:
The gopro hero4 stream received by Periscope is 432 x 240. This is not 720p.
Resizing it to 720 during re-encoding on phone or on servers dont improve the low quality.

KonradIT, 2016-06-23:
@sfarges oh well, that is quite embarrasing then

KonradIT, 2016-06-23:
Should I close the issue?

dvize, 2016-07-08:
Makes sense to close since it is not feasible.  Now i know i have to get a capture card.

closed 2016-07-09 by KonradIT