KonradIT/goprowifihack #24: Hero 4 Session: Automatically enable app mode / wifi on startup?

collaxo, 2016-04-12:
Hi Konrad,

do you know if there is any possibility to enable the app mode / wifi mode on startup after a cold start? We are using the Hero 4 Session in an customized underwater housing with a wifi antenna inside and are not able to press buttons to manually switch to the app mode.

Thank you in advance for your support!

Best regards
Patrick

KonradIT, 2016-04-18:
Hi, that might be doable using cal.txt but I am not sure yet. Other solution would be to turn on WiFi before putting it in the housing, the WiFi in the HERO4 Session does not suck so much battery while the camera is dormant

silent3rror, 2016-04-21:
I think the only way it's possible is by using override.sh and using echo command
`echo 138 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio138/direction
echo 1 > /sys/class/gpio/gpio138/value
sleep 3
echo 0 > /sys/class/gpio/gpio138/value
echo in > /sys/class/gpio/gpio138/direction
echo 138 > /sys/class/gpio/unexport`

collaxo, 2016-04-23:
Hi, thank you for your answers. I tried some of the autoexec-hacks but with no success. What have i do to get the override.sh hack working? (We are using the latest firmware of the Hero 4 Session)

As you mentioned (@KonradIT), our current approach is to turn on WiFi before putting the camera in its housing. Based on your reverse engineering efforts with the WiFi REST interface, we developed our own driver to control the Hero 4 Session. The camera is integrated with a Liquid Robotics Wave Glider and therefore it should be working for weeks and months. The camera is powered by USB while it's in the housing, so we do not have problems with the energy. The main problem comes from unexpected firmware freezes, which force us to shutdown the external power until the camera powers down itself. After enabling the external power again, the camera should start in app mode, so we are able to establish a new connection via the REST interface.. 

Any further thoughts on that?

KonradIT, 2016-04-23:
Mmmm... cal.txt should definitively do it, but I could not get cal.txt hacks to work on the latest HERO4 Session Firmware (released quite recently) v2.00

Let me see what I can do

silent3rror, 2016-04-23:
Here are some cal.txt commands to test, IF you can get cal.txt to work. I don't have a HERO4 Session, so I am unfortunately unable to test these commands.
[https://github.com/hypoxic/hero4-session/blob/master/t_app_test.txt](url)
[https://github.com/hypoxic/hero4-session/blob/master/t_app.txt](url)
[https://github.com/hypoxic/hero4-session/blob/master/t.txt](url)

collaxo, 2016-04-23:
The latest firmware fixed some pretty annoying stuff (like the time to respond and power up) but if we could get the command scripting via cal.txt to work again, we should give it a try and downgrade the firmware to v1.50. The GoPro support website tells that downgrading the camera will "destroy" it.. Do you have any expericences with downgrading?

http://gopro.com/help/articles/Question_Answer/Downgrading-Your-Camera-Software

silent3rror, 2016-04-23:
I personally do not have experience downgrading, I think it would work, because if it worked before the upgrade, why wouldn't it work from downgrading..But don't take my word on that..I feel like GoPro might be telling people that, so that people would be scared to downgrade for the exact reason of cal.txt and autoexec.....but here https://www.youtube.com/watch?v=9gDSOy42Nfs  It shows a downgrade (not session though, but I imagine if Gopro says it will damage all cameras, but it's doable on one, then it would probably work on the session)

silent3rror, 2016-04-24:
Confirmed working downgrade from v03.00.00 to v01.02.00 on HERO4 Silver, and my camera isn't broken..but I can't seem to get autoexec.ash to work like here https://vimeo.com/110696475 (@KonradIT can you spread some light on that?) So like I said above, I don't think you'll break the cam, but it'll definitely void your warranty. And as for override.sh, I am afraid that the special firmware is only for HERO3/HERO4 (Black/Silver). If you already have a modded firmware, try these steps: 

1. Extract telnet.zip to root of SD card (make sure the two files are not in a subfolder when extracted)
2. Put the SD card back into the cam and power it on
3. (If the wifi comes on automatically, thats good, if not don't worry about it)Wait for about 15 seconds and telnet to 10.5.5.9
4.Telnet is automatically installed on Mac and Linux, but disabled in windows but to enable it is a pain, so if using windows, use a program called putty (mirror download link: https://the.earth.li/~sgtatham/putty/latest/x86/puttytel.exe )
5. If using Mac or Linux, open terminal and enter `telnet 10.5.5.9` or using putty, put 10.5.5.9 into the hostname/IP address input and click open (make sure telnet is selected)

6. If override.sh works, you will get a BusyBox root shell. If you get an error, that probably means that it isn't going to work.
[telnet.zip](https://github.com/KonradIT/goprowifihack/files/233209/telnet.zip)

KonradIT, 2016-04-24:
@collaxo Hi, I think I got it for WiFi on boot, please tell me how are you turning the camera ON? I have success with cal.txt making adjustments to the lens (https://www.instagram.com/p/BEjpAnXnitS) and some _tapp command can turn WiFi on, my procedure was: hit the button on the back, wait until it goes to "more" and then it sleeps, and then cal.txt should kick, because right after the camera goes to sleep the LEDs turn on, a coincidence that the cal.txt takes exatcly 7 seconds to start and the GoPro HERO4 Session takes 7 seconds from start to sleep, you can also start the camera to record instantly using the shutter button and cal.txt *should* kick after 7 seconds

@silent3rror forget about override.sh in the HERO4 Session, the HERO4 Session uses cal.txt (I finally got cal.txt to work in my HERO4 Session running 2.00 !!!) also for override.sh to work it needs a modded FW made by Alti Force, and they dont support the HERO4 Session for obvious reasons.

The video you linked was demoing the capabilities of autoexec.ash on a HERO4 Black with a in-house prototype firmware that I got access to, but autoexec.ash support never made it to the public releases.

To get a Linux USB shell working in the H4 Session: https://github.com/hypoxic/hero4-session/tree/master/linux_terminal

collaxo, 2016-04-27:
@KonradIT Hi, i'm turning the camera on by pressing the small button on the back side of the gopro session. Then i'm using this button to navigate to the app menu and confirm the app mode by pressing the record button on the top side.

Can you provide a simple cal.txt file with the capability of turning the wifi on? Thank you in advance! And good luck for your exams! ;-)

gutmanu, 2016-11-29:
Hi, I´ve just read through your conversation. Unfortunately it´s just not clear to me if there was a solution in the end. To solve my problem I only need the wifi running after 5V are supplied. That feature worked with firmware 1.5 but does´t do it any more with 2.0 . 

I would also downgrade the camera, but where to get the old firmware?

Thanks a lot for your support!

collaxo, 2016-11-29:
There is still no solution. We finally switched to another camera.

closed 2017-01-21 by KonradIT

maxabernathy, 2018-01-09:
Hi, I'm sorry to come back to this topic but I have the same camera and would like to use it with an autonomous usb supply, so no problem with energy sparsity but there's still the wifi issue as it will not be accessible. If cal.txt is not a solution, is there another camera or way that would do the trick? Thanks for your work!