KonradIT/goprowifihack #25: Hero 4 Black - change in Address

ed-george, 2016-04-20:
I created my [own guide](https://github.com/ed-george/GoPro4Streaming) before realising this repo was available.

I noted today that in firmware `HD4.02.01.02.01`, the new stream URL to access is [http://10.5.5.9/gp/gpExec?p1=gpStreamA9&c1=restart](http://10.5.5.9/gp/gpExec?p1=gpStreamA9&c1=restart)

This should probably be reflected in the documentation 👍

KonradIT, 2016-04-23:
Hi, I am based all my API research on the latest firmware for all cameras, currently the HERO4 Black and Silver latest firmware update is **HD4.02.03.00.00** from July 30, 2015 as time of writing. I recommend you update your camera not only to get performance improvements and new features but also to maintain compatibility with this API.

closed 2016-04-23 by KonradIT

ed-george, 2016-04-24:
Thanks for the tip, I wasn't aware the firmware hadn't been updated. The option to update wasn't immediately available within the software, but it is now. Thanks