KonradIT/goprowifihack #32: Announcement: Pausing development due to exams

KonradIT, 2016-04-26:
Hello contributors,

Next week I start my IGCSE exams I will not be able to do much coding/internet activities, etc... So, until June I might not be very active here.

Things in my TODO list:

* Make a resolution/framerate table [relevant](https://github.com/KonradIT/goprowifihack/issues/18)
* Get the Trimming video URLs
* Get the 720p streaming method (URL, UDP call, who knows?) iOS device needed
* Get HERO4 Silver resolution/framerate values from /gp/gpControl
* Karma drone research (wifi/RC???)

If you have time consider doing these and answer GH issues.

Best,

Konrad

silent3rror, 2016-04-28:
@KonradIT What do you mean by HERO4 Silver res/frame values? Most if not all are the same as the HERO4 Black. I think that I know what your asking..I'm working on updating gp-control..and here's what the values are in the HTML file.
**NOTE:** This includes NTSC/PAL frame rates 
`<tr><td>Video resolution HERO4 Silver</td><td>
  <button onclick="command('gp','gpControl','setting','2','1')">4K</button>
  <button onclick="command('gp','gpControl','setting','2','2')">4K SuperView</button>
  <button onclick="command('gp','gpControl','setting','2','4')">2.7k</button>
  <button onclick="command('gp','gpControl','setting','2','5')">2.7k SuperView</button>
  <button onclick="command('gp','gpControl','setting','2','6')">2.7k 4:3</button>
  <button onclick="command('gp','gpControl','setting','2','7')">1440p</button>
  <button onclick="command('gp','gpControl','setting','2','8')">1080 SuperView</button>
  <button onclick="command('gp','gpControl','setting','2','9')">1080</button>
  <button onclick="command('gp','gpControl','setting','2','10')">960p</button>
  <button onclick="command('gp','gpControl','setting','2','11')">720 SuperView</button>
  <button onclick="command('gp','gpControl','setting','2','12')">720p</button>
  <button onclick="command('gp','gpControl','setting','2','13')">WVGA</button></td></tr>
<tr><td>Frame Rate</td><td>
<button onclick="command('gp','gpControl','setting','3','0')">240fps</button>
<button onclick="command('gp','gpControl','setting','3','1')">120fps</button>
<button onclick="command('gp','gpControl','setting','3','2')">100fps</button>
<button onclick="command('gp','gpControl','setting','3','3')">90fps</button>
<button onclick="command('gp','gpControl','setting','3','4')">80fps</button>
<button onclick="command('gp','gpControl','setting','3','5')">60fps</button>
<button onclick="command('gp','gpControl','setting','3','6')">50fps</button>
<button onclick="command('gp','gpControl','setting','3','7')">48fps</button>
<button onclick="command('gp','gpControl','setting','3','8')">30fps</button>
<button onclick="command('gp','gpControl','setting','3','9')">25fps</button>
<button onclick="command('gp','gpControl','setting','3','10')">24fps</button>
<button onclick="command('gp','gpControl','setting','3','11')">15fps</button>
<button onclick="command('gp','gpControl','setting','3','12')">12.5fps</button></td></tr>`

KonradIT, 2016-04-29:
@silent3rror thanks, will merge. Can you do the resolution/frame rate table? First bullet point, just check what franerates are available for each resolution

silent3rror, 2016-04-29:
@KonradIT I have a conference to go to tonight and tomorrow, I will get to it on Sunday.

silent3rror, 2016-04-29:
@KonradIT If I have any free time tonight, i'll definitely get to it

KonradIT, 2016-04-29:
@silent3rror I will be available only today so I will do Hero4 Black/Hero4 Session/Hero3 ALL models Frame rate/resolutions table now. After tomorrow its studying all day for me

KonradIT, 2016-04-29:
@silent3rror also push the framerate/resolutions table to Framerate-Resolutions.md file in /HERO4/ (will make file now!)

silent3rror, 2016-04-29:
@KonradIT I have a table already for HERO4 Black/session/silver I'm just formatting it now

silent3rror, 2016-04-29:
Actually then because I don't know all of the markdown, just take my repo patch-6 and format it.. I'll create a pull request and you can just rename the file

KonradIT, 2016-04-29:
Ok, push it and I will fix it! Also, for MD tables have a look at this: http://www.tablesgenerator.com/markdown_tables

KonradIT, 2016-04-29:
HERO4 Framerate/resolution table done in fcf509d (@silent3rror and @konradit)

closed 2016-05-26 by KonradIT