KonradIT/goprowifihack #40: Wifi disconnect every time

RoToRx88, 2016-05-30:
####Problem:
Everything works well on my GoPro HERO3 I can start/stop the GoPro and recording etc... But the wifi connection disconnect automaticly after a few seconds I am connected, and I just have the time to send 4/5 commands.
Moreover, when I connect my PC (archlinux) to the gopro's wifi I have to perform a `dhclient wlo1` if I want an IP.

Do you now how can I fix or if someone else had the same problem ?

Thanks;

####Details:

* GoPro Camera(s): HERO3
* Firmware Version:
* Steps to reproduce: Connect with a linux to wifi; perform a `dhclient` then wait a few seconds (~30sec)
* Happens every time? [Y/N]: Y

silent3rror, 2016-05-31:
> Moreover, when I connect my PC (archlinux) to the gopro's wifi I have to perform a `dhclient wlo1` if I want an IP.

I'm not quite sure what you're meaning by "if I want an IP" 
Running `nmap` or `ifconfig` should give you an ip

As for reproducing this problem, I tried running `dhclient` but with an error as follows: 
`root@kali:~# dhclient
Job for smbd.service invalid.
invoke-rc.d: initscript smbd, action "reload" failed.`

silent3rror, 2016-05-31:
After restarting `smbd` I came out with this
`root@kali:~# dhclient
RTNETLINK answers: File exists`

KonradIT, 2016-06-01:
Will test this afternoon

KonradIT, 2016-06-06:
Strange, I can connect flawlessly to my GoPro H4 from my laptop running Arch and GNOME. I connect via the GUI Settings application

silent3rror, 2016-06-06:
Same here running Kali and GNOME, I connect via the GUI aswell. @RoToRx88 please reproduce the problem, then paste the output of `dmesg`

RoToRx88, 2016-06-10:
Hi,

I'm back with the dmesg log...
So tried to connect to the GoPro in wifi. Connected, then run `dhclient wlo1` to get an IP (otherwise I do not get any IP) then just wait and get disconnected.
I don't really understand, the dmesg log says that it is a _local choice_ to disconnect... Why ?

Here is it:
```
[ 4321.597944] wlo1: authenticate with d8:96:85:23:19:a3
[ 4321.643141] wlo1: send auth to d8:96:85:23:19:a3 (try 1/3)
[ 4321.646220] wlo1: authenticated
[ 4321.647972] wlo1: associate with d8:96:85:23:19:a3 (try 1/3)
[ 4321.654382] wlo1: RX AssocResp from d8:96:85:23:19:a3 (capab=0x431 status=0 aid=1)
[ 4321.675961] wlo1: associated
[ 4321.678960] cfg80211: Regulatory domain changed to country: US
[ 4321.678964] cfg80211:  DFS Master region: FCC
[ 4321.678966] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[ 4321.678969] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 3000 mBm), (N/A)
[ 4321.678971] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 1700 mBm), (N/A)
[ 4321.678973] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2300 mBm), (0 s)
[ 4321.678975] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2300 mBm), (0 s)
[ 4321.678976] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 3000 mBm), (N/A)
[ 4321.678978] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[ 4368.148753] wlo1: deauthenticating from d8:96:85:23:19:a3 by local choice (Reason: 3=DEAUTH_LEAVING)
[ 4368.169161] cfg80211: World regulatory domain updated:
[ 4368.169164] cfg80211:  DFS Master region: unset
[ 4368.169165] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[ 4368.169167] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[ 4368.169169] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[ 4368.169170] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[ 4368.169172] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[ 4368.169174] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[ 4368.169175] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[ 4368.169176] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[ 4368.169178] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
```

I'm looking on archlinux forum too to understand the problem !

(In parallel I tried to connect a raspberry Pi 2 in wifi to my gopro without success on raspbian...
Maybe I'll switch to archlinux later, I'll keep you informed.

silent3rror, 2016-06-10:
@RoToRx88 Did you update the Kernel recently? Also try these two things:

1. Paste the output of `lspci`

2. Try running
 `sudo iw reg set US` and `sudo sed -i 's/^REG.*=$/&US/' /etc/default/crda` as suggested in these two forums (Yes, I know this is for Ubuntu, but this looks like it's a kernel issue) http://ubuntuforums.org/showthread.php?t=2259037
http://superuser.com/questions/974017/ubuntu-wlan0-authenticates-and-then-drops

RoToRx88, 2016-06-10:
@silent3rror I can't find the file `/etc/defualt/crda` but I investigated in `systemctl` for the service `NetworkManager` and here is the output when I am connected on the GoPro and I can execute command from `curl` or `wgeŧ` on the gopro IP etc...

```juin 10 16:21:20 rotor-pc NetworkManager[482]: <info>  [1465568480.2361] sup-iface[0xc130a0,wlo1]: config: set interface ap_scan to 1
juin 10 16:21:20 rotor-pc NetworkManager[482]: <info>  [1465568480.2524] device (wlo1): supplicant interface state: disconnected -> scanning
juin 10 16:21:23 rotor-pc NetworkManager[482]: <info>  [1465568483.1442] device (wlo1): supplicant interface state: scanning -> authenticating
juin 10 16:21:23 rotor-pc NetworkManager[482]: <info>  [1465568483.1471] device (wlo1): supplicant interface state: authenticating -> associating
juin 10 16:21:23 rotor-pc NetworkManager[482]: <info>  [1465568483.1718] device (wlo1): supplicant interface state: associating -> associated
juin 10 16:21:24 rotor-pc NetworkManager[482]: <info>  [1465568484.1701] device (wlo1): supplicant interface state: associated -> 4-way handshake
juin 10 16:21:24 rotor-pc NetworkManager[482]: <info>  [1465568484.1981] device (wlo1): supplicant interface state: 4-way handshake -> completed
juin 10 16:21:24 rotor-pc NetworkManager[482]: [159B blob data]
juin 10 16:21:24 rotor-pc NetworkManager[482]: <info>  [1465568484.1984] device (wlo1): state change: config -> ip-config (reason 'none') [50 70 0]
juin 10 16:21:24 rotor-pc NetworkManager[482]: <info>  [1465568484.1990] dhcp4 (wlo1): activation: beginning transaction (timeout in 45 seconds)
```

And here is the output of `systemctl status NetworkManager` when I am disconnected after a few seconds (in facts 45 seconds each time I try)

```
juin 10 16:21:24 rotor-pc NetworkManager[482]: <info>  [1465568484.1990] dhcp4 (wlo1): activation: beginning transaction (timeout in 45 seconds)
juin 10 16:22:09 rotor-pc NetworkManager[482]: <warn>  [1465568529.7931] dhcp4 (wlo1): request timed out
juin 10 16:22:09 rotor-pc NetworkManager[482]: <info>  [1465568529.7932] dhcp4 (wlo1): state changed unknown -> timeout
juin 10 16:22:09 rotor-pc NetworkManager[482]: <info>  [1465568529.8051] dhcp4 (wlo1): canceled DHCP transaction
juin 10 16:22:09 rotor-pc NetworkManager[482]: <info>  [1465568529.8052] dhcp4 (wlo1): state changed timeout -> done
juin 10 16:22:09 rotor-pc NetworkManager[482]: <info>  [1465568529.8059] device (wlo1): state change: ip-config -> failed (reason 'ip-config-unavailable') [70 120 5]
juin 10 16:22:09 rotor-pc NetworkManager[482]: <warn>  [1465568529.8072] device (wlo1): Activation: failed for connection 'GOPRO-BP-D896852319aÿ'
juin 10 16:22:09 rotor-pc NetworkManager[482]: <info>  [1465568529.8088] device (wlo1): state change: failed -> disconnected (reason 'none') [120 30 0]
juin 10 16:22:09 rotor-pc NetworkManager[482]: <warn>  [1465568529.8191] sup-iface[0xc130a0,wlo1]: connection disconnected (reason -3)
juin 10 16:22:09 rotor-pc NetworkManager[482]: <info>  [1465568529.8199] device (wlo1): supplicant interface state: completed -> disconnected
```

We can clearly see that it is the dhcp4 that disconnect me from the GoPro.

It is odd because I perform a `sudo dhclient wlo1` to obtain an IP and then I HAVE an IP... Otherwise I should not be able to do any actions on the GoPro with `curl` etc...
But it seems that the `dhcp4` started by `NetworkManager` does not take care of my manual start of a `dhclient`...
I am now trying to set my IP statically but I still have the same problem / comportment...

(Sorry for my English I'm french ^^) If you need more details you're welcome !

I'm keep working on the problem.

silent3rror, 2016-06-10:
@RoToRx88 Try here (set IPv6 to link local) https://bbs.archlinux.org/viewtopic.php?id=146261
I'll keep looking up forums and finding possible solutions.

RoToRx88, 2016-06-10:
It does not works :(