KonradIT/goprowifihack #41: Command to trigger LRV creation of a video?

awidgery, 2016-06-09:
Do we know the command to trigger the camera to create an LRV for a specific video?

KonradIT, 2016-06-09:
GoPro HERO2, HERO3, HERO4 and HERO+/LCD generate a .LRV video if the video is shot in: 4K 30/24, 2.7K 60/30/24, 1080p 60/30/24, 720p 120/60/30, WVGA 240/120 with **No ProTune enabled**

closed 2016-06-09 by KonradIT

awidgery, 2016-06-13:
Thanks - it doesn't look like .LRV files are generated automatically for 720p 240fps videos.

I'm trying to intercept the requests between the iOS app and the GoPro by using Charles Proxy, but I'm not having any luck. What tools do you use to intercept requests/packets to do your reverse-engineering?

KonradIT, 2016-06-17:
@awidgery In the HERO2/3 days I used Wireshark and tcpdump with success, but with the new generation cameras (HERO4 Black, Silver, Session) both Wireshark and tcpdump dont show any HTTP GET commands from the GoPro App, I also tried installing Shark app on my rooted phone but no cigar, so now I extract the commands from the GoPro App APK