KonradIT/goprowifihack #46: Undocumented settings values on status page

networkjanitor, 2016-07-04:
####Problem:

There seem to be some undocumented settings values on the status page of a gopro (http://10.5.5.9/gp/gpControl/status)

In the settings object the following key=value pairs could not be identified (neither with the help of this documentation nor with the information gathered from the gpControl page of a gopro http://10.5.5.9/gp/gpControl)

Undocumented Key,Value pair: 40 = 0
Undocumented Key,Value pair: 41 = 13
Undocumented Key,Value pair: 42 = 8
Undocumented Key,Value pair: 43 = 0
Undocumented Key,Value pair: 44 = 8
Undocumented Key,Value pair: 45 = 8
Undocumented Key,Value pair: 46 = 0
Undocumented Key,Value pair: 47 = 0
Undocumented Key,Value pair: 48 = 0
Undocumented Key,Value pair: 65 = 0
Undocumented Key,Value pair: 66 = 0
Undocumented Key,Value pair: 67 = 0
Undocumented Key,Value pair: 71 = 0

Any ideas what these could be ?

####Details:

* GoPro Camera(s): HERO4 Silver
* Firmware version: HD4.01.03.00.00
* Steps to reproduce: http://10.5.5.9/gp/gpControl/status
* Happens every time? [Y/N]: Y

KonradIT, 2016-07-04:
What object? settings or status?

networkjanitor, 2016-07-04:
settings

KonradIT, 2016-07-04:
I'll see what my HERO4 Session gives, I'll be travelling from the 7th to 20th and the HERO4 Black arrives the 14th so until the 20th for HERO4 Black results

KonradIT, 2016-07-04:
Oh and could you test these URLs with your HERO4 Silver? Fetched from a friend's camera with old firmware so it might not work:

* LCD ON: http://10.5.5.9/gp/gpControl/setting/72/1
* LCD OFF: http://10.5.5.9/gp/gpControl/setting/72/0

* LCD Never sleep: http://10.5.5.9/gp/gpControl/setting/51/0
* LCD 1min sleep timeout: http://10.5.5.9/gp/gpControl/setting/51/1
* LCD 2min sleep timeout: http://10.5.5.9/gp/gpControl/setting/51/2
* LCD 3min sleep timeout: http://10.5.5.9/gp/gpControl/setting/513

networkjanitor, 2016-07-04:
All of them work and LCD ON/OFF is already documented.

networkjanitor, 2016-07-04:
Just saw something as I was closing tabs: The gopro actually returns an error code when setting the LCD sleep setting, but the value changes in the gopro settings (on the display, aswell as in the settings object).

KonradIT, 2016-07-04:
Value 40 is date time (0 must be set not set from app?), Value 41 is OTA status (I don't know what 13 is supposed to mean here), Value 43 is current mode (see CameraStatus, this was used as a command value in old firmwares), other values seem to be related with internet streaming feature (nor publicly available yet)

networkjanitor, 2016-07-04:
Are you sure about this ? Because the meanings/usages you named are for the status object and not for the settings object.

networkjanitor, 2016-07-04:
40 seems to accept values [0, 1]
48 seems to accept values [0, 1, 2]
47, 67  seem to accept any values (only tested numeric)

All other (undocumented) settings seem to immutable.

rosek86, 2016-12-24:
This is from Android app decompilation:

40 - BROADCAST_DEFAULT_SUB_MODE
41 - BROADCAST_RESOLUTION
42 - BROADCAST_FPS
43 - BROADCAST_FOV
44 - BROADCAST_RECORD_RESOLUTION
45 - BROADCAST_RECORD_FPS
46 - BROADCAST_RECORD_FOV
47 - BROADCAST_WINDOW_SIZE
48 - BROADCAST_PRIVACY
65 - BROADCAST_GOP_SIZE
66 - BROADCAST_IDR_INTERVAL
67 - BROADCAST_BIT_RATE
71 - BROADCAST_CURRENT_SUB_MODE

It seems like another recording mode, I'm not sure what it does for now..

KonradIT, 2016-12-24:
@rosek86 I'm guessing that is either an unreleased streaming feature that "accidentally" (as always) made it to production or commands to use with GoPro's HEROCast.

closed 2017-01-21 by KonradIT