KonradIT/goprowifihack #47: Hero 4 Session LiveStream in v02.00

ivanmkc, 2016-07-20:
####Problem:
Livestream GET request has no effect.

Side-note: I would like to help investigate this. What's the best way to set up the packet sniffing on Android to do that? Thanks!

####Details:
I sent the GET request: 
`http://10.5.5.9/gp/gpControl/execute?p1=gpStream&a1=proto_v2&c1=restart`

I get the response:
`{status: "0"}`

Live stream mode does not start. I can confirm this by no change on the Hero 4 Session Display, as well as no UDP packet stream for video.

* GoPro Camera(s):
Hero 4 Session

* Firmware Version:
v02.00

* Steps to reproduce:
I sent the GET request: 
`http://10.5.5.9/gp/gpControl/execute?p1=gpStream&a1=proto_v2&c1=restart`

* Happens every time? [Y/N]:
Y

KonradIT, 2016-07-20:
Hi @ivanmkc, use the GoPro App to make sure your camera is up to date. I recently played around with a HERO4 Silver and used the URL you mentioned to get a live feed using FFplay.

ivanmkc, 2016-07-20:
Thanks for the quick response @KonradIT. It is brand new and up-to-date.

From the official site (https://gopro.com/update/hero4_session), the Hero 4 Session runs the HERO4 Session v02.00 firmware (March 31, 2016) and the Hero 4 Black\Silver run firmware v03.00.00 (July 30, 2015). Given the 8 month difference in release, I'm not surprised that there are discrepancies between them.

I'd love your help resolving this, but I don't mind helping either if you tell me how I can best contribute.

Thanks,
Ivan

ivanmkc, 2016-07-20:
For reference here is the output from: http://10.5.5.9/gp/gpControl

The GET request looks to be as your guide says them to be, but the GoPro4 Session just doesn't react to the command.

`{
	"version": 4.0,
	"display_hints": [{
		"key": "GPCAMERA_GROUP_VIDEO",
		"display_name": "Video Settings",
		"settings": [{
			"setting_id": 5,
			"widget_type": "select",
			"precedence": 1
		}, {
			"setting_id": 6,
			"widget_type": "select",
			"precedence": 1
		}, {
			"setting_id": 2,
			"widget_type": "select",
			"precedence": 1
		}, {
			"setting_id": 3,
			"widget_type": "select",
			"precedence": 1
		}, {
			"setting_id": 4,
			"widget_type": "select",
			"precedence": 1
		}, {
			"setting_id": 8,
			"widget_type": "select",
			"precedence": 1
		}, {
			"setting_id": 9,
			"widget_type": "toggle",
			"precedence": 1
		}, {
			"setting_id": 10,
			"widget_type": "toggle",
			"precedence": 1
		}, {
			"setting_id": 11,
			"widget_type": "select",
			"precedence": 1
		}, {
			"setting_id": 12,
			"widget_type": "select",
			"precedence": 1
		}, {
			"setting_id": 13,
			"widget_type": "select",
			"precedence": 1
		}, {
			"setting_id": 14,
			"widget_type": "select",
			"precedence": 1
		}, {
			"setting_id": 15,
			"widget_type": "slider",
			"precedence": 1
		}],
		"commands": [{
			"command_key": "GPCAMERA_VIDEO_PROTUNE_RESET_TO_DEFAULT",
			"precedence": 1
		}]
	}, {
		"key": "GPCAMERA_GROUP_PHOTO",
		"display_name": "Photo Settings",
		"settings": [{
[... 1806 lines cut ...]
			"url": "/gp/gpPlatformAuth"
		},
		"fw_update": {
			"version": 1,
			"description": "Supports client-assisted Over-the-Air firmware updating.",
			"url": "/gp/gpUpdate"
		},
		"supports_app_clipping": {
			"version": 1,
			"description": "Supports mobile app clipping.",
			"url": ""
		}
	},
	"info": {
		"model_number": 16,
		"model_name": "HERO4 Session",
		"firmware_version": "HX1.01.02.00",
		"serial_number": "C3141325552495",
		"board_type": "0x07",
		"ap_mac": "F4DD9E1FB503",
		"ap_ssid": "blackthunder",
		"ap_has_default_credentials": "0",
		"git_sha1": "1f8812f42c9dfefd38360a02308f2992fb55fe77"
	}
}`

KonradIT, 2016-07-20:
@ivanmkc crap I read it as HERO4 **Silver**, I am aware of this problem as I have a HERO4 Session myself and cannot get to stream. I tried using Charles Proxy but didn't get anything relevant. I have the Session with me now, will see what I can do

ivanmkc, 2016-07-20:
@KonradIT Ya I picked the session as it's form factor makes it great for mounting on a drone. I think I read that you had interest in drones as well ;)

How do you set up Charles proxy to view the Android traffic? I used Wireshark on my Mac (on promiscuous mode) and couldn't see it. Would love to help.

KonradIT, 2016-07-20:
If your Android phone is rooted, try Termux or Terminal Emulator with tcpdump or mitmdump, that should help. Or decompile the GoPro smarty apk

KonradIT, 2016-07-20:
OK I know what's happening here (I guess). My phone being pity does not let me run tcpdump/mitmdump because of some PIE setting. I used "Packet Capture" app from the Play Store and got some interesting requests. The app is wonderful: no root needed, material design...

Screenshots here:

http://i.imgur.com/VviiGhB.png

http://i.imgur.com/bmLKIRX.png

KonradIT, 2016-07-20:
Forgot to mention /gp/gpControl/camera/PV/2 gives 500 Internal Error. /camera/PV/2 is used in HERO2 and HERO3 cameras but not in HERO4, I think they just messed up (being GoPro, it does not surprise me at all)

ivanmkc, 2016-07-20:
I used the app like you suggested. It shows me what you screenshotted.

So is the answer to send a preview ON request after sending the stream request?

KonradIT, 2016-07-20:
I will need to reproduce everything in the right order later

ivanmkc, 2016-07-20:
Ok, let me know when you do and I'll test it out. Thanks!

KonradIT, 2016-07-23:
@ivanmkc have you gotten anything? just got home so I'll see if I can get a working stream

KonradIT, 2016-07-23:
I'd take a bet and say that maybe they keep preview alive by using /gp/gpControl/status every second?

KonradIT, 2016-07-23:
GoPro app also sends outdated or bad requests, such as the pairing success and the /camera/PV/2 from HERO3... why?

ivanmkc, 2016-07-24:
Ya, I wonder if the outdated requests are actually being used for some
side-effect that keeps the connection alive. I'm bogged down with work this
week but I'll try this out later on in the week.

Ivan

KonradIT, 2016-07-25:
Got the preview working in FFplay after using the GoPro App on mobile. They still use udp 8554, although I didn't check any variable status in /gp/gpControl/status

https://i.sli.mg/FEq7IG.png

All its missing is a preview ON url

ivanmkc, 2016-07-25:
Ah I should have  mentioned that was what I was doing to get it working in
the meantime.
Seems like the stream is working fine as you say.

chengsam, 2016-08-04:
Hi guys, I am trying to get my Session to stream to an iOS app. I am able to stream for now but encounters some difficulties for it to stream easily. I am not able to stream immediately after turning the wifi of GoPro on but after I tried visiting gp/gpControl/status for several times it can start to stream. I don't know if this is the magic method to allow it to start streaming. Also, the python script I used sends the "_GPHD_:0:0:2:0.00000" command to keep it alive. Can I do it also on iOS? Thank you.

KonradIT, 2016-08-04:
>gp/gpControl/status for several times

I had my doubts regarding this but since you got it working I'll update the repo accordingly.

The python sends a magic packet with that command, same as GoPro App, you need to find a library for ObjC that does that, [such as this one](https://github.com/trusk89/wol_lib-master-iOS9) (Not tested with Hero4 Session!)

I'll confirm later on that the stream works with Hero4 Session on FFplay

chengsam, 2016-08-05:
Your doc mentioned that the magic packet is as:

> send a magic packet (WoL) with these parameters: MAC ADDRESS OF THE CAMERA, 10.5.5.9 as IP ADDRESS, Subnet Mask 255.255.255.0, Port 9.

But the python script sent "_GPHD_:0:0:2:0.00000", are they the same?

KonradIT, 2016-08-05:
@chengsam No they are not the same, I think that keeps the stream alive where as the magic packet turns on the camera

chengsam, 2016-08-05:
OK, I can stream the preview to my app now. However, the delay is ~5s. Just figuring out how to reduce the latency.

KonradIT, 2016-08-05:
@chengsam didn't have time yesterday I was travelling, did you send /gp/gpControl/status GET request to start the stream? Are you on 2.00?

chengsam, 2016-08-05:
Yes, I sent /gp/gpControl/status several times until the status 31 of the camera and it's not 0 to start the stream. But I'm not sure if it works all the time. And my version is 2.00.

KonradIT, 2016-08-06:
Can confirm it works with pinging /gp/gpControl/status 5 times. I only neeeded ti HTTP GET the URL 5 times at the start and then run the stream on FFplay. I used [this script](https://github.com/Sonof8Bits/GoProStreaming/blob/master/GoProStream.py) and it worked. I'll push a PR to add HERO4 Session support and update the docs on this repo accordingly.

Proof: http://imgur.com/a/PHtoi

Closing thread.

closed 2016-08-06 by KonradIT

chengsam, 2016-08-11:
I found that sometimes it requires me to ping more than 5 times to make status 31 > 0. So in my implementation, i check for the count of status 31 and start the stream when it's not 0.

KonradIT, 2016-08-11:
@chengsam I'll write it then, thanks!