KonradIT/goprowifihack #51: Command to finalize pairing process

zaur, 2016-08-24:
One more thing to add about pairing :)
Actually we need to run 2 commands in order to finish pairing process (at least on HERO4).
First is
https://10.5.5.9/gpPair?c=start&pin=XXXX&mode=0
And right after that finish pairing:
https://10.5.5.9/gpPair?c=finish&pin=XXXX&mode=0

For me PIN is always 4 digits (seen on Hero4).

####Problem:

####Details:

* GoPro Camera(s): HERO4
* Firmware Version: 4.0.0
* Steps to reproduce:
* Happens every time? [Y/N]: Y

KonradIT, 2016-08-24:
Thanks, added in f35c448254de23ea315059fcadc341e585a780d3

closed 2016-08-24 by KonradIT

dengyunjie, 2018-01-23:
Hi, I send this URL : https://10.5.5.9/gpPair?c=start&pin=XXXX&mode=0 .GoPro returns “ BT: RECV: HTTP/1.1 400 Bad Request ，Connection: close” 。Have you met this problem?