KonradIT/goprowifihack #53: HERO4 Looks like MDNS Protocol

cmagorian, 2016-11-04:
####Problem:

From intercepting the WIFI calls with wireshark, I'm seeing only MDNS protocols, not http protocols on the IPV4 stack. None of the commands you have listed work, am I missing some key piece here?

####Details:

Can't figure out HTTP protocols when sniffing the WIFI traffic

* GoPro Camera(s):
Hero 4 Black
* Firmware Version:
v5.0
* Steps to reproduce:
Sniffing traffic from a Hero 4 black to GoPro App
* Happens every time? [Y/N]:
Y

KonradIT, 2016-11-04:
Hi, I used "Packet Capture" app from Play Store and it did sniff the commands, since it can sniff thru SSL. I did test most of the commands with my HERO4 Black w/ 5.00 firmware, what did you try?

cmagorian, 2016-11-04:
I have the same firmware and hardware. I tried both tcpdump abd Wireshark, but did not have SSL cert on phone or laptop "mitm", maybe this was my issue. Weird that it reads as UDP though with SSL

cmagorian, 2016-11-06:
Another question:

How would one emulate the "pairing" process? I haven't been able to find an endpoint on the same config with packet capture that can pair with the camera.

Thanks!
Chris

KonradIT, 2016-11-06:
@cmagorian https://github.com/KonradIT/goprowifihack/blob/master/HERO4/WifiCommands.md#code-pairing

closed 2017-01-21 by KonradIT