KonradIT/goprowifihack #57: Browsing not working on Hero 5 Black

auxiliary, 2016-12-08:
####Problem:
It seems like the /videos url doesn't work on the Hero 5. When I navigate to the URL I get the following error: 
```
Error 404: Not Found
Not found
```
####Details:

* GoPro Camera(s): Hero 5 Black
* Firmware Version: 1.55
* Steps to reproduce: Connect to the hero's wifi and navigate to http://10.5.5.9:8080/videos
* Happens every time? [Y/N]: Yes

KonradIT, 2016-12-08:
Hi @auxiliary 

Please go to http://10.5.5.9:8080/videos/DCIM/ and see if you can browse!

You can also get the file-list here: http://10.5.5.9:8080/gp/gpMediaList

Spoiler: I actually don't have a HERO5 camera, all I gathered was from a friend's camera, before 1.55 FW hit. Let me know if something is broken!

auxiliary, 2016-12-09:
Thank you. [http://10.5.5.9:8080/videos/DCIM/](http://10.5.5.9:8080/videos/DCIM/) works. I had searched the whole net and couldn't find the solution. You're a lifesaver.

KonradIT, 2016-12-09:
Added to the HERO5 page. Closing.

closed 2016-12-09 by KonradIT

mikinet1, 2019-02-05:
Problem: 

I have Hero 7 silver,
and the "http://10.5.5.9:8080/gp/gpMediaList", is not working ,

I get  the response:
"{protocol=http/1.1, code=200, message=OK, url=http://10.5.5.9:8080/gp/gpMediaList} "

how do i get the media list?
Thanks.

KonradIT, 2019-02-05:
Hey @mikinet1 , I don't have the HERO7 Silver, only the Black one. Mind opening a new issue where we can continue this? I haven't gone into the HERO7 White or Silver API functions yet but you can help me get the calls.

mikinet1, 2019-02-05:
no problem

it's open HERO7 Silver #140

ardianumam, 2019-06-10:
Hi, 
After connecting via WIFI, I cannot browse both http://10.5.5.9:8080/videos/DCIM/  and http://10.5.5.9:8080/gp/gpMediaList. I use goPro 5 (black). FYI, I can run "ping 10.5.5.100" and "ping 10.5.5.9" successfully. Any idea?

asd864613087, 2020-07-17:
It seems that, in Hero 7 Black, this url doesn't work now - http://10.5.5.9:8080/gp/gpMediaList (error code : 503).  But this one is Ok - http://10.5.5.9:8080/videos/DCIM/ .