KonradIT/goprowifihack #58: GOPRO OMNI Firmare?

roysbike, 2016-12-14:
####Problem:
I need firmware OMNI  for gopro4 
####Details:

* GoPro Camera(s):
* Firmware Version:
* Steps to reproduce:
* Happens every time? [Y/N]:

KonradIT, 2016-12-14:
Hi,

This repo is dedicated to documenting and reverse engineering the **wireless** protocol that GoPro HERO cameras use to communicate with the handheld Capture app. OMNI VR rig is out of the scope.

With firmware, do you mean the firmware on the camera, because the HERO4 updated to the latest firmware supports OMNI's HEROBUS protocol AFAIK, if you mean the firmware that is on the connector bus or other electrical component then I don't know since I don't have or have access to an OMNI VR Rig.

roysbike, 2016-12-14:
I access omni firmware OmniSecondaryCamera https://yadi.sk/d/1Pxc2xy533Xpr9
And OmniPrimaryCameraFirmware https://yadi.sk/d/Pn5SDSpx33XwEg. You may need to.

roysbike, 2016-12-14:
I want make analog omni rig. But 6 cameras are collected in an array. Synchronisation is via the bus gopro. But switching circuit I do not have. How do you think it is possible to know the connection type of connector? If there is a firmware omni?

KonradIT, 2016-12-15:
You should ask GoPro Support center, I don't have this kind of information.

closed 2016-12-15 by KonradIT