KonradIT/goprowifihack #61: Hero5

shayanb, 2017-01-21:
I noticed there's no API details for Hero5? Is there one under development or not started yet?

KonradIT, 2017-01-21:
Hey @shayanb , to put it briefly the HERO5 cameras share the same WiFi methods as the HERO4 cameras, including live streaming preview, file handling and wifi HTTP GET commands. The HERO5 cameras have features that the HERO4 does not (such as EIS, Voice control, WDR, etc...). The commands for the new features are documented [here](https://github.com/KonradIT/goprowifihack/tree/master/HERO5).

I got this originally from a friend's H5B camera, I got one myself but had to send it back because of extreme light bleeding and unresponsiveness, now I am waiting for the [Karma+HERO5 Session](http://shop.gopro.com/EMEA/karma).

The missing stuff is [in the scrum board](https://github.com/KonradIT/goprowifihack/projects/1).

closed 2017-01-21 by KonradIT

shayanb, 2017-01-21:
@KonradIT nice, it was a bit confusing there but now I got it.

Great it seems like most of the HTTP calls are documented. I'd probably work on a python API just to start playing with some automation. 

Let me know if you need to test anything on Hero5.

KonradIT, 2017-01-22:
@shayanb can you pull /gp/gpControl and /gp/gpControl/status on fw v1.55? I think some API calls change. Also, I have a python script that shows a preview: http://github.com/konradit/goprostream, but a overall Python API would be great, similar to [DenisCarriere/gopro](https://github.com/DenisCarriere/gopro).

If you can manage to [get some of this stuff](https://medium.com/@konrad_it/hacking-gopro-hero5-part-3-f213648c883c#.cfu9ca1z3) working it would be great, as it brings scripting abilities to HERO5.