KonradIT/goprowifihack #62: Hero 5 Livestreaming

SokolovRobert, 2017-01-29:
####Problem:
Can't livestream with Hero5 to computer. Have followed instructions [here](https://github.com/KonradIT/goprowifihack/blob/master/HERO4/Livestreaming.md) along with many other references. I'm not sure if this is due to a firmware update, but on wireshark I'm not seeing any udp traffic whatsoever.

####Details:
Would like to wirelessly stream from Hero 5 to computer.

* GoPro Camera(s): Hero 5 Black
* Firmware Version: HD5.02.01.55.00
* Steps to reproduce:
* Happens every time? [Y/N]: Yes

KonradIT, 2017-01-29:
Try this python script: http://github.com/konradit/goprostream

It only works on Linux/Mac atm.

SokolovRobert, 2017-02-03:
On Linux I still get errors with my Hero 5:

python GoProStream.py 
Traceback (most recent call last):
  File "GoProStream.py", line 112, in <module>
    gopro_live()
  File "GoProStream.py", line 48, in gopro_live
    response = urllib2.urlopen('http://10.5.5.9/camera/cv').read()
  File "/usr/lib/python2.7/urllib2.py", line 154, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python2.7/urllib2.py", line 435, in open
    response = meth(req, response)
  File "/usr/lib/python2.7/urllib2.py", line 548, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python2.7/urllib2.py", line 473, in error
    return self._call_chain(*args)
  File "/usr/lib/python2.7/urllib2.py", line 407, in _call_chain
    result = func(*args)
  File "/usr/lib/python2.7/urllib2.py", line 556, in http_error_default
    raise HTTPError(req.get_full_url(), code, msg, hdrs, fp)
urllib2.HTTPError: HTTP Error 404: Not Found

It's connected properly and I can browse the video content through a browser, but still not achieve any form of stream to the computer.

KonradIT, 2017-02-03:
@SokolovRobert thanks for the log, fixing now.

KonradIT, 2017-02-03:
@SokolovRobert should be fixed now. Please report back.

SokolovRobert, 2017-02-04:
Looks like it's working now but, but ffplay isn't starting properly? No errors, but also no preview terminal comes up, and when trying to open ffplay in a new window the ports are already bound.

zappy@ubuntu:~/Downloads/GoProStream-master$ python GoProStream.py 
HERO4/HERO5/HERO+ camera
('UDP target IP:', '10.5.5.9')
('UDP target port:', 8554)
('message:', '_GPHD_:0:0:2:0.000000\n')
Recording on camera: False
Press ctrl+C to quit this application.

I'm running Ubuntu 16.04 LTS and can check on a Mac later this evening.

KonradIT, 2017-02-04:
@SokolovRobert set the VERBOSE flag to True in the python script, let's see what's wrong

udayakumarka, 2017-05-03:
I tried the script and nothing works.
Is there any guide for live streaming to windows?

KonradIT, 2017-05-03:
@udayakumarka streaming does not work on Windows

udayakumarka, 2017-05-03:
@KonradIT Thanks for the quick response. I was hoping there should be some workaround atleast. With the same way the Capture App is showing the live previews. Hope it will be available for future versions.

KonradIT, 2017-05-03:
@udayakumarka it works in Linux (arch tested so far)

udayakumarka, 2017-05-04:
@KonradIT Any reason why it doesn't work with Windows?

KonradIT, 2017-05-04:
@udayakumarka https://github.com/KonradIT/GoProStream/issues/3

TUDarmstadtLLX, 2017-05-04:
Hello, can I get the Live Stream on windows system with gopro sessiocn 5?

KonradIT, 2017-05-04:
@TUDarmstadtLLX, @udayakumarka FFmpeg does not work on windows for reasons unknown yet, so try the ffmpegless-opencv.py [here](https://github.com/KonradIT/gopro-py-api/blob/master/examples/opencv_gopro/ffmpegless-preview.py)

So, the way this works is by using the OpenCV python library to display the preview directly from the camera without doing ffmpeg > localhost > opencv (this is very stable but I guess it does not work on windows). To try it, install the gopro-py-api by downloading the repo and doing ```python setup.py install``` (you may need admin rights) or if you have ```pip``` installed (usually in the ```[python dir]/tools/pip.exe``` run ```pip install goprocam```. Next is to install OpenCV in your computer, follow [this link]() to download the installer. Then set your HERO5 to WiFi App mode and connect your computer to the wireless network, launch a python instance and type:
```from goprocam import GoProCamera, constants
gopro_hero5 = GoProCamera.GoPro()
```

and wait until it says connected to 10.5.5.9 (the camera should turn on). It should take some time to interface with the camera and turn it on, if after 30 seconds it did not turn on run the ```gopro_hero5 = GoProCamera.GoPro()``` a second time. If even after two times it does not work the [see this thread](https://github.com/KonradIT/gopro-py-api/issues/4#issuecomment-297953690) and test the scripts there (it should help me to see which WOL config turns the camera on or off). If you're lucky and the camera turns on, then run this in your python terminal:

```gopro_hero5.overview()```

It should display some data from the HERO5 including mode, battery, etc... Check the clients connected value and note it down.

Then close the python terminal and run the ffmpegless-preview.py file. It should display the preview, might lag or might not. 

NB: I don't have a HERO5 camera and I don't have Windows, so I will need logs, screenshots, videos, anything that helps me debug. Gotta study now.

TUDarmstadtLLX, 2017-05-04:
Hi, thanks for your relay.

is it possible that I just control the gopro session 5 on the window system? Such as start the video or start to capture the images.

I want to use use LabVIEW as the develepment enviroment. In LabVIEW, there are modules for the using of "http".

KonradIT, 2017-05-04:
Hi @TUDarmstadtLLX you can indeed start and stop recording or take photos. 

```gopro_hero5.take_photo()``` takes a picture

```gopro_hero5.shoot_video()``` starts a video

KonradIT, 2017-05-04:
I've been trying to get the stream to work on Windows, the UDP packets from the GoPro will not be received at all. this is definitely a Windows issue as neither FFmpeg, ODB or OpenCV have gotten to preview the stream

TUDarmstadtLLX, 2017-05-05:
Hi,
what is the command for the taking a picture or video. I have tried the command, which is listet [here](https://github.com/KonradIT/goprowifihack/blob/master/HERO4/WifiCommands.md), but failed to do that.

KonradIT, 2017-05-05:
@TUDarmstadtLLX did you turn the camera on via the gopro.power_on() command

TUDarmstadtLLX, 2017-05-05:
I turned it on manually. But I tried the command for "Power off", it works. How can I do that with the command?

KonradIT, 2017-05-05:
use the GoPro python API, it's based on goprowifihack. You can use the aforementioned command to turn on.

KonradIT, 2017-05-05:
@TUDarmstadtLLX

TUDarmstadtLLX, 2017-05-05:
Hi, @KonradIT, thanks for your reply. If I do not use the python API, I can not take pictures and shot videos via the command, is it true?
Sorry for the question. I have no experience with neither java nor python.

KonradIT, 2017-05-05:
@TUDarmstadtLLX you can take pictures and record videos with the command line using the python api. look at the videos in the README

TUDarmstadtLLX, 2017-05-05:
@KonradIT , ok, thanks!

KonradIT, 2017-05-05:
@udayakumarka 

VirtualBox might not be able to receive UDP packets, even though I edited the firewall config on Windows.

[image]

(Right terminal: opencv running on arch, while windows inside VM cannot open the stream from opencv or ffmpeg)

TUDarmstadtLLX, 2017-05-08:
I am using the window system. If I want to get the Live Stream and be able to set up the camera, such as set the FPS, I need to install the linux system on my PC and use the Python API. Is it true?

Sorry for such foolish question

TUDarmstadtLLX, 2017-05-08:
Hi, I am using a GoPro 5 Session.

TUDarmstadtLLX, 2017-05-08:
@KonradIT Hi, I just thought , for my research, I do not need to get the live stream. I only need to control this GoPro 5 session wirelessly on my computer, such as set up the camera, take pictures and shot videos.

If there are any http commands with which I can do that, that would be awesome.

KonradIT, 2017-05-08:
Hi @TUDarmstadtLLX you can use the GoPro Python API to control the camera (take pictures and shoot videos) even set the FPS.

TUDarmstadtLLX, 2017-05-10:
@KonradIT  Thanks, where is the Python API, Could you post the link?

TUDarmstadtLLX, 2017-05-10:
I have tried the Python API for the GoPro 4, but it returns error. I do not know, if this API passes the GoPro

TUDarmstadtLLX, 2017-05-10:
I have tried the Python API on the Linux for the GoPro 4, but it returns error. I do not know, if this API passes the GoPro 5 Session.

KonradIT, 2017-05-10:
@TUDarmstadtLLX what error? post some logs otherwise I cannot fix anything, also, what HERO4 model do you have?

TUDarmstadtLLX, 2017-05-11:
@KonradIT I have solved the problem. Just reset the GoPro. Great API.

TUDarmstadtLLX, 2017-05-14:
@KonradIT  Hi, I just run the GoPro Cam API under windows system, I send a command "gopro_hero5.shoot_video(3)", the camera implements the command and shoots a video for 3 seconds, but I returns the error like the following picture
[image]

KonradIT, 2017-05-14:
hey @TUDarmstadtLLX thanks for the log! I'll test it on windows and fix it.

KonradIT, 2017-05-14:
@TUDarmstadtLLX run this

```gopro_hero5.getStatus(constants.Status.Status, constants.Status.STATUS.IsBusy)``` and post the output here.

TUDarmstadtLLX, 2017-05-30:
Hi, my last GoPro Session 5 is broken. I do not know why. I have sent a command to it, it suddenly turns off and can not power on anymore.

So I exchanged it with the new GoPro Black 4. I have got the live stream under windows system work. But there is about 2 second delay. Is ist normal?

TUDarmstadtLLX, 2017-05-30:
@KonradIT Hi, my last GoPro Session 5 is broken. I do not know why. I have sent a command to it, it suddenly turns off and can not power on anymore.

So I exchanged it with the new GoPro Black 4. I have got the live stream under windows system work. But there is about 2 second delay. Is ist normal?

KonradIT, 2017-05-30:
i haven't gotten the streaming to work on windows yet so I can't really say.

LanceRagas, 2017-06-10:
@KonradIT Hi, I have read the comments here and am having the same issue. I am running Windows on my pc and have had no success at all. I am using a Hero Session for my live streaming. Because of this, would the live stream still work?

KonradIT, 2017-06-10:
I could get it to work on windows @LanceRagas

LanceRagas, 2017-06-11:
@KonradIT When I run the python script, it says the machine is actively refusing it. Would this be solved with send a WoL?

LanceRagas, 2017-06-11:
@TUDarmstadtLLX How did you get your gopro to connect through windows?

TUDarmstadtLLX, 2017-06-11:
@LanceRagas I can connet my GoPro Black 4 under the mode of APP. Before I have also used a GoPro Session 5, it can also connect with windows under the APP mode. Just need to give the wifi password.

KonradIT, 2017-06-11:
@TUDarmstadtLLX what version of Windows? can you run the stream successfully?

LanceRagas, 2017-06-11:
@TUDarmstadtLLX I have everything setup. I am sending a WoL to my Session and it is recording without the SD card. But the same error occurs. Would this be due to my computer running Windows 7?

TUDarmstadtLLX, 2017-06-11:
@KonradIT  I have testet it on Windows 8.1. I can get livestream with VLC and ffmpeg.

TUDarmstadtLLX, 2017-06-11:
@LanceRagas Sorry, I have not testet it on the Windows 7. When I test the livestream with my GoPro session 5, it is not only unstable, the camera is finally broken under the control of the Python API. I have no idea of the reason.

LanceRagas, 2017-06-16:
@TUDarmstadtLLX to get the camera to work windows 10, what would be the correct process to follow?

KonradIT, 2017-08-13:
@TUDarmstadtLLX Hi, can I get your email?

TUDarmstadtLLX, 2017-08-13:
Hi,

my E-Mail ist xixiliebeweiwei@gmail.com

Best regard

nitinveludandi, 2017-12-06:
@KonradIT I am using a Mac and I have hero5 session, I cannot get the live streaming to work. It doesn't give me any error but just after giving me all the camera and flag details its stuck. there is no UDP window to show the preview. Same is the case with Hero4 silver as well. can you help.