KonradIT/goprowifihack #65: HERO5 Turn on does not work

KonradIT, 2017-02-18:
####Problem:

The Wake On Lan command does not work with GoPro HERO5 Black on latest FW version 1.57

Fix needed ASAP

####Details:

* GoPro Camera(s): GoPro HERO5 Black
* Firmware Version: 1.57
* Steps to reproduce: Run wake on lan 10.5.5.9 port 9
* Happens every time? [Y/N]: Y

gokseltur, 2017-02-21:
Hi,
I have test this issue HERO5 Black, Hero5 Session and Hero4 Black last firmware. Cameras opening but doesn't respond communicate requests.

KonradIT, 2017-02-21:
I can turn on my HERO4 Black with a Wake On Lan command, not HERO5.

Teo47, 2017-02-21:
Wake on lan command? What IS it?

KonradIT, 2017-02-21:
@Teo47 that's a standard protocol for turning on devices connected to a network: https://en.wikipedia.org/wiki/Wake-on-LAN

To turn on your camera use this code:

    wol -i 10.5.5.9 -p 9 XX:XX:XX:XX:XX:XX

XX:XX:XX:XX:XX:XX is the MAC address of the camera.

gokseltur, 2017-02-21:
@KonradIT i seen 118 byte in packet. not 128. I think related that ..

{"info":{"model_number":19,"model_name":"HERO5 Black","firmware_version":"HD5.02.01.57.00","serial_number":"C316132XXXXXXX","board_type":"0x05","ap_mac":"f6dd9e44e47a","ap_ssid":"GP246XXXXX","ap_has_default_credentials":"0","capabilities":"16"}}

[image]

KonradIT, 2017-02-22:
@gokseltur mmmm...

KonradIT, 2017-03-23:
[@arthurpattee got it working on his H5 Session with wakeonlan command](https://github.com/KonradIT/gopro-py-api/issues/2#issuecomment-288222434)

KonradIT, 2017-04-08:
Hey @gokseltur try the API now, it should have fixed the issue.

closed 2017-04-12 by KonradIT