KonradIT/goprowifihack #66: Hero+ bad request

singha24, 2017-03-07:
####Problem:
Hi, I have connected my laptop via wifi to my hero+, and running the python script using python3

####Details:

File "GoProStream.py", line 112, in <module>
    gopro_live()
  File "GoProStream.py", line 47, in gopro_live
    response = urllib.request.urlopen('http://10.5.5.9/gp/gpControl/info').read()
  File "/usr/lib/python3.5/urllib/request.py", line 163, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.5/urllib/request.py", line 472, in open
    response = meth(req, response)
  File "/usr/lib/python3.5/urllib/request.py", line 582, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python3.5/urllib/request.py", line 510, in error
    return self._call_chain(*args)
  File "/usr/lib/python3.5/urllib/request.py", line 444, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.5/urllib/request.py", line 590, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 400: Bad Request

Also if I hit http://10.5.5.9/gp/gpControl/execute?p1=gpStream&a1=proto_v2&c1=restart in my browser the return JSON is {"status":"0"}

* GoPro Camera(s): Hero+
* Firmware Version: 01.50
* Steps to reproduce:
python3 GoProStream.py
* Happens every time? [Y/N]: Y

KonradIT, 2017-03-07:
Thanks I'll look into it.

singha24, 2017-03-08:
> Thanks I'll look into it.

Awesome! Thank you, really appreciate it!!

KonradIT, 2017-03-15:
Hey, try again with the new code.

singha24, 2017-03-16:
Hey @KonradIT , thanks for taking the time to look into it... however I am unable to get a stream.. the output my terminal is as follows:

HERO+
HD3.22.01.50
UDP target IP: 10.5.5.9
UDP target port: 8554
message: _GPHD_:0:0:2:0.000000

Recording on camera: False
Press ctrl+C to quit this application.

ffplay version 2.8.11-0ubuntu0.16.04.1 Copyright (c) 2003-2017 the FFmpeg developers
  built with gcc 5.4.0 (Ubuntu 5.4.0-6ubuntu1~16.04.4) 20160609
  configuration: --prefix=/usr --extra-version=0ubuntu0.16.04.1 --build-suffix=-ffmpeg --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --cc=cc --cxx=g++ --enable-gpl --enable-shared --disable-stripping --disable-decoder=libopenjpeg --disable-decoder=libschroedinger --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmodplug --enable-libmp3lame --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librtmp --enable-libschroedinger --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzvbi --enable-openal --enable-opengl --enable-x11grab --enable-libdc1394 --enable-libiec61883 --enable-libzmq --enable-frei0r --enable-libx264 --enable-libopencv
  libavutil      54. 31.100 / 54. 31.100
  libavcodec     56. 60.100 / 56. 60.100
  libavformat    56. 40.101 / 56. 40.101
  libavdevice    56.  4.100 / 56.  4.100
  libavfilter     5. 40.101 /  5. 40.101
  libavresample   2.  1.  0 /  2.  1.  0
  libswscale      3.  1.101 /  3.  1.101
  libswresample   1.  2.101 /  1.  2.101
  libpostproc    53.  3.100 / 53.  3.100
    nan    :  0.000 fd=   0 aq=    0KB vq=    0KB sq=    0B f=0/0

KonradIT, 2017-03-16:
That's strange. 

What are your script configs?

jplpereira, 2020-04-10:
I'm having the same issue with GoPro Hero 3+