KonradIT/goprowifihack #67: Take a single picture

matt-the-bass, 2017-03-09:
####Problem:
I'd like to be able to take a single picture from a GoPro Hero 5.  Ideally, I'd like to use a single URL to take a new picture and get that picture. This is similar to how Axis network cameras work.

Is this possible? I understand how to configure the picture settings and I've been able to browse a listing of pictures but not trigger a new picture and capture it.

####Details:

* GoPro Camera(s): Hero 5
* Firmware Version: ?
* Steps to reproduce: n/a
* Happens every time? [Y/N]: Y

KonradIT, 2017-03-10:
Hi,

My python library supports that: http://github.com/konradit/gopro-py-api

gopro.take_photo() takes a photo and returns the URL

matt-the-bass, 2017-03-10:
Thanks @KonradIT for the quick reply. I found this project of yours. I should be able to use this in a simple python script to meet my needs. I appreciate your help!

closed 2017-03-10 by matt-the-bass