KonradIT/goprowifihack #68: Question: Internet Access

LoganArnett, 2017-03-13:
#### Question
Has anyone integrated this into an application that also needs access to the internet? I am trying to figure out how to remote control the GoPro, allow the user to take a photo or video, and then upload the files to AWS s3 to be displayed on a custom gallery landing page.

Any ideas would be awesome, thanks

* GoPro Camera(s): Hero 5 Black

KonradIT, 2017-03-13:
You could use a raspberry pi if you figure out how to connect to an ethernet connection and a wifi connection at the same time, or, have the rpi constantly waiting for a command connected to internet wifi and then switch to gopro, take the photo, download it, switch to internet wifi, then upload it to s3.

I did something similar years ago: https://github.com/konradit/gopro-upload

LoganArnett, 2017-03-13:
Interesting, thanks

ollietb, 2017-10-03:
If you're on Windows then you can connect another wi-fi adapter and have one connected to the gopro and one connected to the internet. Windows supports multiple wifi adapters out of the box.

closed 2017-10-04 by LoganArnett