KonradIT/goprowifihack #70: HERO5 Firmware Update 2.00

KonradIT, 2017-04-08:
The new update introduces [new protune controls for photo and multishot](https://twitter.com/David_Newman/status/849723785892712448)

Here is the changelog:

**Highlights:**

  * **Added new languages for camera and voice control**
  * **Applied existing settings to more capture modes**
  * **Added GoPro Plus features**

**Usability Improvements**

  * Adds Korean, Russian, and Portuguese (Brazil) as options for camera and voice control languages*
  * Adds setting for date and time format, based on language selection 
  * Includes touch display optimizations
  * Adds ability to extract a photo from a multi-shot series

**Performance Improvements**

  * Adds Shutter setting for Photo capture mode in Protune™ (options are AUTO, 1/125, 1/250, 1/500, 1/1000, 1/2000 second)
  * Adds Wide Dynamic Range (WDR) for Time Lapse Photo (intervals of at least 5 seconds)
  * Adds RAW Format for Time Lapse Photo and Night Lapse Photo (intervals of at least 5 seconds)
  * Adds RAW Format for Night photo

**GoPro Plus Improvements**

  * Allows manual upload of selected content

--- 

The command for 4K 24FPS for HERO5 Session should be the same as HERO5 Black. 

Needed: 

* shutter setting for Photo capture
* WDR for TimeLapse
* RAW format for TimeLapse/NightLapse
* RAW format for NightPhoto 
* Adds ability to extract a photo from a multi-shot series (?)

B3nD3lan3y, 2017-04-26:
Has anyone experienced wake-over-LAN issues after this update? I'm working with some Hero 5 Sessions, updated one, and am now having issues waking it up.

KonradIT, 2017-04-26:
@B3nD3lan3y any error logs?

KonradIT, 2017-06-02:
5a4161f973d9c467fcad98656143d93ed69e59e9

closed 2017-06-02 by KonradIT