KonradIT/goprowifihack #77: Control over bluetooth

realies, 2017-07-06:
Have you explored the option of controlling GoPro cameras over bluetooth? Is it actually possible? The [information](https://www.reddit.com/r/gopro/comments/2isk3d/gopro_hero_4_bluetooth_connectivity_only_works/) I'm finding suggests this is only possible on iOS to pair the devices, after which the camera is controlled via WiFi. I'm not willing to believe they've wasted the bluetooth radio for that. Thoughts?

KonradIT, 2017-07-06:
http://developer.gopro.com/info they also allow camera control using BLE but right now there isn't a device that supports it. I already posted the commands to pair with Bluetooth devices.

realies, 2017-07-06:
@KonradIT, planning on connecting via a RPi 3, which [should](http://www.cypress.com/file/298076/download) support BLE (Bluetooth 4.1). I assume there would be a bluetooth specific API and the reverse-engineered http requests would be irrelevant.

KonradIT, 2017-07-06:
Someone needs to get their hands on a BLE compatible GoPro controller device, which there isn't right now

realies, 2017-07-06:
Thought that smartphones have BLE for years now. Maybe decompiling the camera firmware could reveal more hints... :o)

KonradIT, 2017-07-06:
maybe, you want to try?

realies, 2017-07-07:
@KonradIT, sure, @evilwombat has some tools for decompiling the firmware, can keep this thread updated if you're interested.

KonradIT, 2017-07-07:
please do.