KonradIT/goprowifihack #78: Open source lib to control multiple cameras (like Remote Control)?

hb0, 2017-07-14:
First of all: Great work you have done here!

**Problem:**

We need to start two HERO4 cameras at the same time via Wifi (time lapse mode to trigger pictures with 1Hz until we send a stop command). The "app mode" creates a hotspot for each camera, switching between these hotspots takes multiple seconds, so I was wondering if there is an open source library which allows us to control multiple cams at the same time using the "Wifi RC" mode like described here: https://gopro.com/support/articles/controlling-individual-cameras-with-the-wi-fi-remote-when-multiple-cameras-are-paired

**Details:**

* GoPro Camera(s): two HERO4

KonradIT, 2017-07-14:
Hi. I tried doing an API for control remote years ago but the commands didn't show up on my analyzer, I also don't have any wifi RC now.

hb0, 2017-07-14:
Thanks for the reply. I also don't have a RC unit right now but might give it a go if I don't find another solution for that problem. GoPro mentions a new feature of their Android Toolkit: "Control multiple cameras via Bluetooth Low Energy" - did you or someone else tried that yet? https://developer.gopro.com/info

KonradIT, 2017-07-14:
Is anybody doing the mobile BLE thing? if an app uses it let me know and I'll have a look at it. the RC uses Wifi, it creates an AP and the cameras connect to said AP. then the remote sends TCP/IP commands and the cameras reply with a ASCII image of the front panel screen for the remote to display. we could use OpenCV or some other tool to extract the data from the ASCII.

KonradIT, 2017-07-14:
I also sold all my remotes because I use my smartwatch instead of the remote

Quakeman0, 2017-11-03:
I tried the BLE, only really managed to pair the device with the service uuid it gives in a scan and used the "command" char uuid. I tried sending the byte values in to the command uuid but it does nothing, tried a few variations of the same thing and ways of entering but to no avail.
The only small success I've had is getting the bluetooth low energy device to boot up the gopro. All i did to get this to work was setup BLE server with a name, paired the gopro to the BLE server using the smartphone app and set it to autoconnect. For your first time pairing BLE, use the smartphone app pairing process, find and connect to the cameras bluetooth service and you should be able to get the device to power on the camera when they reconnect (may be limited to 8 hours not sure yet...) I have the lcd disabled at the moment and letting it time out would mean my lcd is gone and I can't connect anymore.... :(

Quakeman0, 2017-11-03:
https://gethypoxic.com/blogs/technical/gopro-hero5-interfaces

This guy documents the BLE a bit, but hasn't tried it out

Gorn226, 2017-12-06:
@Quakeman0 I´m also working on this kind of GoPro control, but after more than one week of intensive reasearch and try and error, i didn´t manage to achieve the pairing process or to boot up the gopro :/. Do you have any additional tips, to get this done? I would much appreciate it =).

Quakeman0, 2017-12-06:
> Yeah i will send you what i have in just a bit!

Cheap MCU board with wifi and BT, programmable through usb and has serial
debugging, a hall effect sensor, temp sensor and 12bit precision ADC called
ESP32, NODE32S is a good cheap breakout version of this for interfacing
with a breadboard. Uploading the code to that board will work fine.
0. Upload BLE_servergpro.ino to ESP32 board
1. Pair gopro to smartphone
2. In the gpro app, choose bluetooth connections search for your server
"MyESP32", select it to autoconnect
3. Upload the NODE32sBRUV.ino
4.Start Pairing process for smart phone app on your gopro while the ESP32
board is running
5. Let it do it's magic, should turn Gopro on using the BLE and then
initiate wifi, press shutter and short spamming of sleep mode

(You may need to eliminate all the DS3231 related code (this is just a temp
compensated RTC board that I use with the setup for long accuracy on time
in extreme environments

Quakeman0, 2017-12-06:
If you need, I can remove all the DS3231 code so you can get a better look
at the bluetooth/wifi parts.

Quakeman0, 2017-12-06:
Also during the pairing process, if the ESP32 has already started searching
for the network, it will time out. The gopro pairing setup part has to
happen first before the ESP32 begins its wifi pairing

Quakeman0, 2017-12-06:
I reccomend using BLE Scanner App on the google Play store for
investigating more of the Bluetooth stuff.

SORRY FOR ALL THE EMAILS, I can never think of everything to be written all
in one email aha...

KonradIT, 2017-12-06:
where's *.ino files?

Quakeman0, 2017-12-06:
Should be attachments in the message where I explained the steps

Quakeman0, 2017-12-06:
It's code for the ESP32 boards to use the Arduino IDE

Gorn226, 2017-12-07:
@Quakeman0 Thank you for your fast response =). You mentioned code examples/attachments, but i´m not able to see it.

Quakeman0, 2017-12-07:
I will resend them in the next email you receive

Quakeman0, 2017-12-07:
Here are the .ino files
  Node32sBRUV.ino
<https://drive.google.com/file/d/0B_pgcu1pw9gyU1JFNVRETTY0WWxSRW5mZmJyNlRzOFNJcW53/view?usp=drivesdk>

  BLE_servergpro.ino
<https://drive.google.com/file/d/0B_pgcu1pw9gybnMxMUx1LTlRTHlUUXB1TVd4enRndHhfa084/view?usp=drivesdk>

Gorn226, 2017-12-07:
Thanks mate =). As far as i understand your implementation, you create a server with a service(GoPro) and a characteristic(GoPro Command). Then you start the server and the advertisment.
=> As far as I can see, i did the same procedure, but unfortunately it is not possible to turn on the goPro. Maybe this because i'm using android. (I try to control the goPro via BLE with a smart phone)

hb0, 2017-12-07:
Just wondering: Did any of you guys already try the official "Camera Toolkit for Android" (GoPro Developer Program): https://developer.gopro.com/info?

We were thinking about using this Toolkit to control two GoPros via BLE, i.e.:
- Start and Shut down the cameras
- Trigger Photo Capturing on both cameras with 1-5 Hz 
- Alternatively, trigger video capturing on both cameras
- Optionally, pull the image data from the cameras

Gorn226, 2017-12-07:
@hb0 I didn´t uses that toolkit, because as far as i know, you have to declear/describe on which project you are working. I´m working on a project for a company and due to NDAs i´m not sure, that i'm allowed to share this kind of information. Enlighten me, if i´m wrong about, that you have to commit on which project you are currently working on. Because it would be a huge relief to use this toolkit.

hb0, 2017-12-07:
@Gorn226 You are right, afaik. We had to describe our (also) business product briefly to GoPro because they want to make sure they don't help you to develop a product which they want to offer themselves. In our case it's about collecting vehicle infrastructure data which seemed not to be a problem for them, so they approved our Toolkit-Access. We'll start developing this camera-containing product in Q1-2018, thus, I did not have a closer look into the Toolkit so far.

KonradIT, 2017-12-07:
Relevant slides: https://github.com/quine/GoProGTFO/raw/master/NSC-GoPro_or_GTFO.pdf

Quakeman0, 2017-12-07:
> GoPro_or_GTFO.pdf

For it to auto turn on using BLE you need to have paired that device as
bluetooth and auto connect to it. Those options are in the gopro camera app
for selecting bluetooth device.

Gorn226, 2017-12-08:
@hb0 Thank you, for update, but i´ll have to talk to my supervisor.
@KonradIT I want to thank you as well. At first view, i nearly didn´t understand anything, because neither i have no experience with linux, nor a deeper knowlege of networking/hex dumps. I`ll investigate the paper later with a closer look.

Gorn226, 2017-12-08:
@Quakeman0 Thank for your advise, but this option(auto connect) should be the default setting, after the pairing process. Maybe i misunderstood something, but as far as i understand, if the goPro is switched off, it is not possible for my application/smartphone to connect to it.