KonradIT/goprowifihack #79: HERO 4 Get the filename of the last photo

hsz, 2017-07-19:
Is it possible in any way to take the filename of the last taken photo in HERO 4?

I'd highly like to avoid fetching the big JSON file from http://10.5.5.9:8080/gp/gpMediaList

KonradIT, 2017-07-19:
You could parse http://10.5.5.9:8080/videos/DCIM/[%d]GOPRO/ (HTML cherokee frontend web server site) but I recommend /gp/gpMediaList since its in JSON and easy to parse. You do need to cope with the download time for gpMediaList

hsz, 2017-07-19:
Yeah, that's the problem. After each photo taken, I need to store it's filename and take next one after a less than 1sec (cams connected to the odometer).

Thanks anyway for the quick response!

Good job with all this reverse engineered stuff!

closed 2017-07-19 by hsz

Teo47, 2017-07-20:
If you need take photo with interval less than 1sec it is not possible with the classic mode of this code. You have to use MultiShotMode. Are you working on an MMS if you work with an odometer ? 

Personnally, i have modified a bit the source code.
```
def download_del_MultiShot(self, path=""):
	sequence = []
	ligne = []
	if path == "":
		arr = json.loads(self.listMedia())
		folder = ""
		for i in arr['media']:
			for i2 in i['fs']:
				folder = i['d']
		filename = arr["media"][0]["fs"][len(arr["media"][0]["fs"])-1]["n"]
		lower_bound = arr["media"][0]["fs"][len(arr["media"][0]["fs"])-1]["b"]
		high_bound = arr["media"][0]["fs"][len(arr["media"][0]["fs"])-1]["l"]
		for i in range(int(high_bound) - int(lower_bound)+1):
			f = filename[:4] + str(int(lower_bound) + i) + ".JPG"
			ligne.append(str(f))
			self.downloadMedia(folder, f)
			self.deleteFile(folder,f)
	return ligne
```
I supposed you try to synchronise data of the odometer and the GoPro ?