KonradIT/goprowifihack #82: How to setup gopro hero 4 session without lcd

andyland1, 2017-08-29:
**Wifi connection and setup**

Hey. This is my first message in github so i'm not sure hot this work but i try this way. I have problem with my session 4 lcd screen. The screen totally broken so i try to figure out how to setup the camera settings with another way and i don't have wifi connection to my phone anymore. I know how to reset wifi settings but i can't get the pin code. Is it possible to get that wifi connection to smartphone working without the ldc screen or is it possible to setup another way?

* GoPro Camera(s): Hero 4 session
* Firmware Version:
* Steps to reproduce:
* Happens every time? [Y/N]:

KonradIT, 2017-08-29:
yikes, I don't think you can get the PIN code without the LCD screen. you can set up the configuration using cal.txt : http://github.com/konradit/hero4hack

throttlenerd, 2019-06-11:
Hey! Got the exact same problem. Did you find the way to solve it? )