KonradIT/goprowifihack #83: Android: live streaming from GoPro

chri1389, 2017-09-11:
**Problem:**

I need to develop an Android App that display live streaming from GoPro Hero 4..

Can You help me, please?

I have used a script Python that work very well (it using ffplay), but I do not know how to do with Android.

GoPro Hero4.

KonradIT, 2017-09-11:
FFmpeg-Android from Writing Minds. Learn to use it and adapt GoProStream to udp://127.0.0.1:1000 (see gopro-py-api). Then just play it using your favorite video library (I use Video view)

chri1389, 2017-09-12:
@KonradIT thank you very much!
Can I use this library?
[http://writingminds.github.io/ffmpeg-android-java/](url)

At the end of the page, there is an apk, that I have used only to check if it work.
What is the command to livestream?
I have used:" ffplay -an -fflags nobuffer -f:v mpegts -probesize 8193 rtp://10.5.59:8554" (because it work well from linux terminal)
but the app display "FAILED with output" and "ffplay: invalid argument"

chri1389, 2017-09-12:
@KonradIT  thank you very much!

I am using this library:
[http://writingminds.github.io/ffmpeg-android-java/](url)

and I am watching only the code of this project:
[https://github.com/WritingMinds/ffmpeg-android-java](url)

A question:
what is the command to execute to livestream GoPro Hero4?

I have used this:
`ffmpeg -loglevel panic -fflags nobuffer -f:v mpegts -probesize 8192 udp://127.0.0.1:1000`

because work well from a linux terminal, but the Android App display:

> Unable to find a suitable output format for 'ffplay ... etc..

The execution call more time the "onProcess" method. In my code there is a "Log.i(..." that print the String message parameter in each call-->I have attached a screenshot:
[image]

Can you help me?
Thank you very much!

Best regards!

chri1389, 2017-09-12:
@KonradIT 
I think I'm wrong about the command. What is the right comand?
Thank you.

chri1389, 2017-09-12:
@KonradIT thank you very much!
I have used this library:
[http://writingminds.github.io/ffmpeg-android-java/](url)

This is my code:
`String command = "ffplay -loglevel panic -fflags nobuffer -f:v mpegts -probesize 8192 udp://:8554";`
`String[] cmd = command.split(" ");`
  ...
  `ffmpeg.execute(cmd, new ExecuteBinaryResponseHandler(){` etc...   

I have used this command:
> ffplay -loglevel panic -fflags nobuffer -f:v mpegts -probesize 8192 udp://:8554

because it works well from a linux terminal.

But Android call the `onFailure(String message)` method.

I think that the command is not correct in Android.

Can you help me, please?

Thank you very much!!!
Best Regards!

KonradIT, 2017-09-12:
No. You need to stream udp://:8554 to localhost. See: https://github.com/KonradIT/gopro-py-api/blob/master/goprocam/GoProCamera.py#L784

chri1389, 2017-09-12:
@KonradIT 
Then I should run the Python script on a pc that open a stream udp on its ip (through this command)
`subprocess.Popen("ffmpeg -f mpegts -i udp://" + self.ip_addr + ":8554 -b 800k -r 30 -f mpegts " + addr, shell=True)`

and then the Android library must read this stream from the pc? is it correct?

KonradIT, 2017-09-12:
``` ffmpeg -f mpegts -i udp://" + self.ip_addr + ":8554 -b 800k -r 30 -f mpegts ```

chri1389, 2017-09-12:
@KonradIT 
ok. I must run the python script and use this command:
`ffmpeg -f mpegts -i udp://" + self.ip_addr + ":8554 -b 800k -r 30 -f mpegts`

and then execute the ffpeg library in Android  with this command:
`String command = "ffmpeg udp://:8554";`
`String[] cmd = command.split(" ");`
....
`ffmpeg.execute(cmd, new ExecuteBinaryResponseHandler(){` etc...

Is it correct?

chri1389, 2017-09-13:
@KonradIT 
ok..if I understand it well:

1) I must run the python script in the pc and use this command:
`ffmpeg -f mpegts -i udp://" + self.ip_addr + ":8554 -b 800k -r 30 -f mpegts`

2) then execute the ffpeg library in Android  with this command:
*************************
`String command = "ffmpeg udp://ipComputer:8554";`
`String[] cmd = command.split(" ");`
....
`ffmpeg.execute(cmd, new ExecuteBinaryResponseHandler(){` etc...
*************************

Is it correct @KonradIT ?

chri1389, 2017-09-13:
@KonradIT I am sorry but I am confused.
In my Android App I do an http request to this link:
`http://10.5.5.9/gp/gpControl/execute?p1=gpStream&a1=proto_v2&c1=restart`

and then I execute the FFmpeg Android library. This is the code:
`String command = "ffmpeg -f mpegts -i udp://10.5.5.100:8554 -b 800k -r 30 -f mpegts";`
`String[] cmd = command.split(" ");`
`try {`
`ffmpeg.execute(cmd, new ExecuteBinaryResponseHandler(){`  ...etc...

10.5.5.100 is the ip of the tablet where run the application.
In my code there is a `Log.i(...` in the `onProgress(String message) - onFailure(String message)` methods, that print the value of the "message" parameter. So I have attached a screenshot:
[image]
 
At the end, it print:
`[NULL @ 0xf342a600] Unable to find a suitable output format for 'ffmpeg'`
`ffmpeg: Invalid argument`

What is wrong?
@KonradIT Thank you very much!

chri1389, 2017-09-13:
@KonradIT 
I must use a pc that read the live streaming from GoPro Hero4 and open a new udp stream (and Android read this stream), or can I read the live streaming from gopro directly from Android?

chri1389, 2017-09-14:
Ok thank you very much!
@KonradIT only a question please,
I must use a pc that read the live streaming from GoPro Hero4 and open a new udp stream (and then Android read this stream), or can I read the live streaming from gopro directly from Android?

chri1389, 2017-09-14:
@KonradIT thank you very much!
Meanwhile, I write what I have done...

In my Android App I do an http request to this link (through Volley Library):
`http://10.5.5.9/gp/gpControl/execute?p1=gpStream&a1=proto_v2&c1=restart`
and GoPro respons with this:
`{"status":"0"}`

And then I use FFmpeg Android Library to execute this command:
`String command = "ffmpeg -f mpegts -i udp://"+ip.toString()+":8554 -b 800k -r 30";`
`String[] cmd = command.split(" ");`
`ffmpeg.execute(cmd, new ExecuteBinaryResponseHandler(){` etc........

It cal more times onProgress methd (of the FFmpeg library) and it read for example:
> [h264 @ 0xeecfc000] no frame!
[h264 @ 0xeecfc000] non-existing PPS 0 referenced
Last message repeated 1 times
etc....

and at the end call onFailure method with this messages:

> [NULL @ 0xeecaca00] Unable to find a suitable output format for 'ffmpeg'
                                                                   ffmpeg: Invalid argument

What is wrong?
@KonradIT  thank you

chri1389, 2017-09-15:
Now I can reiceve frames by the GoPro. 
It call always the `onProgress(String message)` method.. 
Message parameter has that values:

`frame=  262 fps= 67 q=29.0 size=      46kB time=00:00:09.00 bitrate=  41.8kbits/s dup=199 drop=0 speed=2.31x`
`frame=  276 fps= 62 q=29.0 size=      60kB time=00:00:09.47 bitrate=  51.9kbits/s dup=200 drop=0 speed=2.12x`
`frame=  296 fps= 59 q=26.0 size=      72kB time=00:00:10.11 bitrate=  58.0kbits/s dup=200 drop=0 speed=   2x`
etc....

Is it correct?
How can I display them?

@KonradIT

chri1389, 2017-09-16:
@KonradIT 

Now I can reiceve frames from the GoPro. 
It call always the `onProgress(String message)` method.. 

"String message" parameter has that values:

`frame=  262 fps= 67 q=29.0 size=      46kB time=00:00:09.00 bitrate=  41.8kbits/s dup=199 drop=0 speed=2.31x`
`frame=  276 fps= 62 q=29.0 size=      60kB time=00:00:09.47 bitrate=  51.9kbits/s dup=200 drop=0 speed=2.12x`
`frame=  296 fps= 59 q=26.0 size=      72kB time=00:00:10.11 bitrate=  58.0kbits/s dup=200 drop=0 speed=   2x`
etc....

Is it correct?
How can I display them, or read the frame content?

@KonradIT

KonradIT, 2017-09-18:
Yes you're receiving data from the GoPro, now stream it to a localhost url (maybe read the docs on ffmpeg to understand how it works) and play the URL back using VideoView/ExoPlayer, etc..

chri1389, 2017-09-19:
@KonradIT thank you very much!! You are very kind!

I have used this command to restream in Android:
`-fflags nobuffer -i udp://:8554 -f rtp -an rtp://10.5.5.102:1000/gopro`

10.5.5.102 is the Ip of a PC where I can see live stream using VLC (through this--> `udp://@:1000/gopro`).

If I stream in localhost url, VideoView write: 

> Can't play this video

Do you know what is the right ffmpeg command?
@KonradIT 

(When I finish, I will publish a tutorial to get live streaming from GoPro on Android)

garydelarosausername, 2017-09-19:
@chri1389 Looking forward to the tutorial! :smiley:

chri1389, 2017-09-19:
@garydelarosausername 
certainly!!
As soon as I can get it to work! ;-)

KonradIT, 2017-09-19:
Ok, let's settle this down.

First, @chri1389 you must understand the mere basics of internet and computer networks, for this I recommend reading [Computer Networks - a top down approach](https://www.amazon.co.uk/Computer-Networking-Top-Down-Approach-International/dp/0273768964) ([available as PDF](http://www.bau.edu.jo/UserPortal/UserProfile/PostsAttach/10617_1870_1.pdf)). 

Android development experience will also help, if you want to leverage your Android developing skills [take a look at EDX Android course](https://www.udemy.com/androidcourse/) or the [coursera Android course](https://www.coursera.org/specializations/android-app-development).

Last but not least it is **critically important** to learn about the tools you're using. @chri1389 you need to look at the man page for ffmpeg (by running `man ffmpeg` on your PC) or by looking at the FFmpeg Wiki: https://ffmpeg.org/documentation.html

Maybe this weekend I can get a PoC rolling, but I doubt it since I'm crushed under a ton of work from school. Please learn so you don't have to depend on me.

chri1389, 2017-10-06:
@KonradIT 

I can display the live stream in Android.

I have used the FFmpeg Android library, and i have used this command:
`String[] cmd = {"-an","-i","udp://:8554", "-f:v", "mpegts","udp://127.0.0.1:8555/gopro"};`

So, I have used the vlc Android library, learning this tutorial:
[http://www.truiton.com/2015/03/stream-rtmp-live-android/](url)

and then, vlc open this url stream: `mFilePath = "udp://@:8555/gopro";`

It display the live stream from GoPro Hero 4, but it has a problem: the video is very bad.
Can I display the video better?

KonradIT, 2017-10-06:
@chri1389 seems like you commented something but I'm not able to see it. You got the preview to work? I tried using ExoPlayer, mPlayer and videoView but could not get it to display. Seems like it needs to be udp > rtmp/m3u8.

chri1389, 2017-10-06:
@KonradIT 
now I can display live stream from GoPro Hero4.

You can use the vlc library (that can display udp stream).
With the ffmpeg library, use this command:
`String[] cmd  = {"-f", "mpegts", "-i", "udp://:8554", "-f", "mpegts","udp://127.0.0.1:8555/gopro?pkt_size=64"};`
and with vlc library, display the stream `"udp://@:8555/gopro"`
and then you can see the live stream.

If you use only vlc library, opening the url "udp://:8554" (without ffmpeg that stream) you can display the live stream, but the quality is very bad...

Now, there is only a problem: there is a delay of 3 seconds...

If you have any question about it, you can ask :)

KonradIT, 2017-10-06:
@chri1389 you mean LibVLC? Haven't tried it, will do tomorrow.

KonradIT, 2017-10-07:
@chri1389 https://github.com/KonradIT/gopro-udp-stream-android

Can't get it to play.

chri1389, 2017-10-07:
Ok @KonradIT!

Try doing these steps:

1) Create a new Android Project

2) Learn this tutorial:
[http://www.truiton.com/2015/03/stream-rtmp-live-android/](url)
and use the "Android VLC SDK" (it is very simple to use, follow the steps).

3) Import and learn about the FFmpeg Android Library.
I have used this command: 
`String[] cmd  = {"-f", "mpegts", "-i", "udp://:8554", "-f", "mpegts","udp://127.0.0.1:8555/gopro?pkt_size=64"};`

4) Set the url that vlc must open:
`mFilePath = "udp://@:8555/gopro";`

5) Very important: you must do a thread that do a http request to this link:
[http://10.5.5.9/gp/gpControl/execute?p1=gpStream&a1=proto_v2&c1=restart](url)
It must do this request, about every seven seconds.
It is important that the first request is after that ffmpeg is started! 

@KonradIT  Let me know if there are problems or it works properly!

KonradIT, 2017-10-07:
@chri1389 thanks, now it works. There's delay, its not smooth and the image has artefacts and turns green, but at least it displays the stream.

chri1389, 2017-10-07:
Perfect @KonradIT
Can we work togheter to take away the delay? :)

KonradIT, 2017-10-07:
@chri1389 sure, needs faster refresh rate so I suggest dump LibVLC and do UDP>x transcoding

chri1389, 2017-10-07:
@konradit ok!
We can write in private and then publish a short tutorial useful for other users!

Frequel, 2017-11-21:
there is the possibility to use WebRTC for a stream on android?

ShyamBarange, 2018-02-06:
@chri1389 
tried all steps you explained , still no success.. :(
can you help me  please?

chri1389, 2018-02-06:
@ShyamBarange sure!

You must follow this steps:
- create a new Android Project
- import and use VLC library. For this, for example you can use this tutorial (use it only to import and use VLC):
[http://www.truiton.com/2015/03/stream-rtmp-live-android/](url)
- you must import FFMPEG library
[http://writingminds.github.io/ffmpeg-android-java/](url)
- now, the FFMPEG command is 
`String[] cmd = {"-f", "mpegts", "-i", "udp://:8554", "-f", "mpegts","udp://127.0.0.1:8555/gopro?pkt_size=64"};`
- the VLC url must be
`mFilePath = "udp://@:8555/gopro";`
- You must also do an http request to the GoPro, to command the Live Streaming.
The request must be about every 7 seconds at this URL:
[http://10.5.5.9/gp/gpControl/execute?p1=gpStream&a1=proto_v2&c1=restart](url)
- It is very important that the first HTPP request is after that ffmpeg is started!
- you must add the INTERNET permission in the AndroidManifest:
`<uses-permission android:name="android.permission.INTERNET" />`
- before run the application, assure that Android Phone is connected to the GoPro WiFi network!

Let me know if there are problems or it works properly!

ShyamBarange, 2018-02-06:
@chri1389 
Thanks for the quick response.
I have imported the VLC library and FFMpeg library successfully and its working (since I can stream video from a URL in VideoView widget in android app). But when my phone is connected to the GoPro WiFi network, it is showing some error message that "unable to create media player".
I might be doing some mistake in passing FFMpeg command. 
I am also doing http request using Volley library. 
still no success. 
can you help me in passing the FFmpeg command.

thanks

chri1389, 2018-02-07:
Hello @ShyamBarange 

Can you write the code?

ShyamBarange, 2018-02-08:
Hey @chri1389 
Here is my MainActivity
Let me know if I am doing something wrong.

`public class HomeActivity extends AppCompatActivity implements IVLCVout.Callback {
    public final static String TAG = "HomeActivity";
    private String mFilePath;
    private SurfaceView mSurface;
    private SurfaceHolder holder;
    private LibVLC libvlc;
    private MediaPlayer mMediaPlayer = null;
    private int mVideoWidth;
    private int mVideoHeight;
    Integer count = 0;

    @Override
    public void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);
        mFilePath = "udp://@:8555/gopro";
        Log.d(TAG, "Playing: " + mFilePath);
        mSurface = (SurfaceView) findViewById(R.id.surface);
        holder = mSurface.getHolder();
    }

    public void onClickPreview(View view) {
        callStream();
        createPlayer(mFilePath);
    }

public void callHTTP() {

        StringRequest request = new StringRequest("http://10.5.5.9/gp/gpControl/execute?p1=gpStream&a1=proto_v2&c1=restart", new Response.Listener<String>() {
            @Override
            public void onResponse(String response) {

                Log.wtf("CODE", response);
            }
        }, new Response.ErrorListener() {
            @Override
            public void onErrorResponse(VolleyError error) {

                Toast.makeText(HomeActivity.this, "Error", Toast.LENGTH_SHORT).show();
            }
        });

        RequestQueue queue = Volley.newRequestQueue(this);
        queue.add(request);

    }

    public void callStream() {
        callHTTP();
        try {
            String[] cmd = {"-f", "mpegts", "-i", "udp://:8554", "-f", "mpegts", "udp://127.0.0.1:8555/gopro?pkt_size=64"};

            FFmpeg ffmpeg = FFmpeg.getInstance(getApplicationContext());
            ffmpeg.execute(cmd, new ExecuteBinaryResponseHandler() {

                @Override
                public void onStart() {
                    count += 1;
                    if (count == 7) {
                        count = 0;
                        callHTTP();
                    }
                }

                @Override
                public void onProgress(String message) {
[... 175 lines cut ...]
    private static class MyPlayerListener implements MediaPlayer.EventListener {
        private WeakReference<HomeActivity> mOwner;

        public MyPlayerListener(HomeActivity owner) {
            mOwner = new WeakReference<>(owner);
        }

        @Override
        public void onEvent(MediaPlayer.Event event) {
            HomeActivity player = mOwner.get();

            switch (event.type) {
                case MediaPlayer.Event.EndReached:
                    Log.d(TAG, "MediaPlayerEndReached");
                    player.releasePlayer();
                    break;
                case MediaPlayer.Event.Playing:
                case MediaPlayer.Event.Paused:
                case MediaPlayer.Event.Stopped:
                default:
                    break;
            }
        }
    }
}`

Massier87, 2018-03-04:
@chri1389  I managed to get a stream following your tutorial, thanks. However, I have a lag in the video stream and it seems to vary:

` 03-04 16:41:08.585  30518-30669/com.gopro.stream D/VLC﹕ [b4463128/77cd] core video output: picture might be displayed late (missing 9 ms)
03-04 16:41:08.999  30518-30518/com.gopro.stream D/STREAM﹕ frame=  530 fps= 35 q=9.2 size=    1751kB time=00:00:20.14 bitrate= 711.9kbits/s speed=1.34x
03-04 16:41:09.086  30518-30663/com.gopro.stream W/VLC﹕ [aeb85728/77c7] core audio output: playback too late (67090): up-sampling
03-04 16:41:09.513  30518-30518/com.gopro.stream D/STREAM﹕ frame=  547 fps= 35 q=11.4 size=    1798kB time=00:00:20.72 bitrate= 710.8kbits/s speed=1.33x
03-04 16:41:10.025  30518-30518/com.gopro.stream D/STREAM﹕ frame=  562 fps= 35 q=11.4 size=    1839kB time=00:00:21.22 bitrate= 709.7kbits/s speed=1.32x
03-04 16:41:10.095  30518-30669/com.gopro.stream W/VLC﹕ [b4463128/77cd] core video output: picture is too late to be displayed (missing 42 ms)
03-04 16:41:10.095  30518-30669/com.gopro.stream D/VLC﹕ [b4463128/77cd] core video output: picture might be displayed late (missing 8 ms)
03-04 16:41:10.565  30518-30518/com.gopro.stream D/STREAM﹕ frame=  578 fps= 35 q=8.9 size=    1891kB time=00:00:21.77 bitrate= 711.2kbits/s speed=1.31x
03-04 16:41:11.075  30518-30518/com.gopro.stream D/STREAM﹕ frame=  594 fps= 35 q=12.2 size=    1934kB time=00:00:22.30 bitrate= 710.3kbits/s speed= 1.3x
03-04 16:41:11.595  30518-30518/com.gopro.stream D/STREAM﹕ frame=  609 fps= 35 q=12.1 size=    1977kB time=00:00:22.81 bitrate= 710.2kbits/s speed=1.29x
03-04 16:41:12.097  30518-30518/com.gopro.stream D/STREAM﹕ frame=  624 fps= 34 q=11.6 size=    2019kB time=00:00:23.29 bitrate= 710.0kbits/s speed=1.28x
03-04 16:41:12.601  30518-30518/com.gopro.stream D/STREAM﹕ frame=  639 fps= 34 q=11.5 size=    2069kB time=00:00:23.79 bitrate= 712.5kbits/s speed=1.28x
03-04 16:41:13.109  30518-30518/com.gopro.stream D/STREAM﹕ frame=  654 fps= 34 q=12.1 size=    2109kB time=00:00:24.29 bitrate= 711.1kbits/s speed=1.27x
03-04 16:41:13.629  30518-30518/com.gopro.stream D/STREAM﹕ frame=  670 fps= 34 q=12.7 size=    2152kB time=00:00:24.82 bitrate= 710.0kbits/s speed=1.26x
03-04 16:41:15.739  30518-30669/com.gopro.stream D/VLC﹕ [b4463128/77cd] core video output: picture might be displayed late (missing 5 ms)`

@KonradIT @chri1389  did you manage to fix the lag? 

If so can I please know how? :)

KonradIT, 2018-03-04:
Hi, 

I haven't managed to fix the Lag yet, I still have to try ExoPlayer and SurfaceView renderer.

ShyamBarange, 2018-03-04:
@Massier87 Hey,
can you please share your code? I tried but I couldn't stream my go pro over android.
here is my mail id: Shyam.barange0311@gmail.com
help will be appreciated.
Thanks.

Massier87, 2018-03-04:
@KonradIT Thanks, Also is it a possibility to replace LibVLC with UDP>x? I couldn't find a PoC...If you can please point to a reference, I can try this path.

@ShyamBarange sent

caarl21, 2018-03-05:
Hello, 
Unfortunately I could not solve the delay.
Now, I'm trying other solutions ...
If you have an idea, please contact me!

chri1389, 2018-03-05:
Hello,
Unfortunately I could not solve the delay.
Now, I'm trying other solutions ...
If you have an idea, please contact me!

Frequel, 2018-03-07:
hello guys, i'm intrested in this project, my objective is receive the video from gopro connected via wi-fi to an andoird device and stream it in broadcast. now the problem is that i didn't find nothing useful , only this issue can help me, as soon as possible i will follow the @chri1389 steps to try something, but in the meanwhile i was thinking that we could make a github together and try to help eachother to develop this code. so if you agree we cn exchange our email adress and organize it all. let me know and thank for your work on this thread :)

KonradIT, 2018-03-07:
@frequel I already have a repository with @chri1389 method; https://github.com/konradit/gopro-udp-android

Frequel, 2018-03-07:
@KonradIT page not found error 404 :/
i'll try to find it on your giuthub page

Frequel, 2018-03-07:
i only found this one https://github.com/KonradIT/gopro-udp-stream-android
is the correct link?

KonradIT, 2018-03-07:
yeah its that one

Frequel, 2018-03-07:
@KonradIT ok thank you, i will try something for myself and as soon as possible i will help on that git.
for opinion exchange we will keep in touch in this issue or in the github? so we can store all information in a single way....

jc999, 2018-03-07:
Hello, I am on this subject also for a few weeks already.
The problem I have is that I can't use libvlc since it's too big for the android device I am trying to get the livestream to (android 4.1.2). And the stock mediaplayer doesn't support udp.
I am now trying to use only exoplayer (release 2 with the UDP fix) to connect directly to the cam but I still get a socket timeout on the udp://:8554... even with the gpStream url , and the keepalive message ( _HDHP_...) sent on udp to the gopro every 2.5 sec... 

Any idea to help me solving this out?

Thank a lot,
JC

KonradIT, 2018-03-08:
@Frequel sure. let's chat over here so everyone can benefit.
@jc999 You might have to transcode to something like M3u8 which Video view can play.

Frequel, 2018-03-09:
hi guys, i´m very noob in andorid studio, so i couldn't do very much these days, i've followed the tutorial linked by @chri1389 and it worked, so now iǘe understood how is working the stream but i have a few problems if you could help me.
1)ffmpeg is imported adding only  compile 'com.writingminds:FFmpegAndroid:0.3.2' in  app/build.gradle ? because in the writingminds guide there are three way and the first one doesn't work for me and the second one, that is in the @KonradIT git which linked me few days ago, in my project doesn't work to. can you explain me what i have to do? i followed other method to import external library in android studio and i always receive an error syncing the buil.gradle file. it could be a my problem or i miss some passages?
2) i cloned the @KonradIT git in this link https://github.com/KonradIT/gopro-udp-stream-android but i can build or run anything, someone that can help me???

i'm learning about android studio and i would like to help in the developing of the git so if you can help me is very appreciated and useful. :)
thank you for your patience

Frequel, 2018-03-09:
i was forgetting a third question, sorry.
3) if i'm not wrong in the previous message you are talking about a lag in the streaming, have you tried to add the nobuffer comand in the ffmpeg line comand? as @KonradIT suggest in this link https://github.com/KonradIT/goprowifihack/blob/master/HERO4/Livestreaming.md

KonradIT, 2018-03-09:
@Frequel you have to make a new project and import the module :app, then use `compile 'com.writingminds:FFmpegAndroid:0.3.2'` to import FFmpeg (you do need to configure Android studio to allow GitHub connections) and then build or sign-build the app.

Frequel, 2018-03-09:
ok thank you, but i have other 2 question
1) the module :app from the archive on the writingminds page, right?
2)in your project the import is already complete? (module imported + compile 'com.writingminds:FFmpegAndroid:0.3.2' )

chri1389, 2018-03-09:
@KonradIT I think that the problem about the delay is VLC media player. 
Now I want to check an other solution: read streaming from FFmpeg Android and restream it to OpenCV Android, and then display it.. What do you think about it?

KonradIT, 2018-03-09:
Sure! Try that, maybe add the `-fflags nobuffer ` and let us know. I will run some tests this afternoon

Sean-Kone, 2018-03-11:
Hello, I'm also trying to stream from GoPro.
I tried the code from https://github.com/KonradIT/gopro-udp-stream-android and it worked for me after some minor changes but the video quality starts ok then degrades drastically and eventually the stream get stuck. 
Do you have any solution for that, maybe some flags/options that needs to be set for FFmpeg?

Also, if / when I get the GoPro stream to the device working good I need the frames, do you know how (or have any suggestion) I can extract the frames while simultaneously displaying the stream on the device??

chri1389, 2018-03-11:
Hello!
We are working to use an other method. Read the stream from Ffmpeg and then restream it in localhost and read from OpenCV. If it works, throught openCv you can extract the frames and display them.

If you are interesting, you can work togheter!

KonradIT, 2018-03-11:
@chri1389 great! Can you share the code online? Would like to try it. How well does OpenCV work on Android?

[34 more replies]