KonradIT/goprowifihack #86: Is this already a GoPro app replacement?

garydelarosausername, 2017-09-19:
It looks like it. Like, everything the GoPro app has, this hack has. What are things missing? I have a Hero5 Black, so what things might be missing on this hack for my camera? I'm looking into this hack right now.

Cool project! Thanks a lot everyone!

KonradIT, 2017-09-19:
This is an API, not a app client. An app replacement I'm working on is http://github.com/konradit/camcontrol (for Android).

No commands or status are missing.

garydelarosausername, 2017-09-19:
Oh ok. What I want is a desktop/laptop/PC version of the GoPro app - like literally all the features of it: Quik, Quikstories, Auto Upload from cam, live preview, etc. I think I saw almost everything the GoPro app has in goprowifihack. Lemme check.....