KonradIT/goprowifihack #87: Record mode in GoProStream.py doesn't run consistently as Preview mode.

garydelarosausername, 2017-09-27:
**Problem:**

I say success ratio is 3:10. It might be that I'm using the [UDP URL in the title - that also says I might have a hard time using it for my work -](https://github.com/KonradIT/goprowifihack/blob/master/HERO4/Livestreaming.md) that's why Record mode isn't consistently working as Preview mode. I run the script like so - python3 *py. My camera just gets stuck recording, the timer don't even move. It's good though that when the Record mode doesn't successfully save the recording, the camera does. 

**Details:**

* GoPro Camera(s): Hero5 Black
* Firmware Version: 02.51
* Steps to reproduce:
1. Go to GoProStream dir.
2. Run script like so: python3 *py
* Happens every time? [Y/N]: Y

KonradIT, 2017-09-27:
Set all your flags to false.

garydelarosausername, 2017-09-27:
That'd be Preview mode then. I want to be able to have my camera record via laptop - Record mode of the script.

KonradIT, 2017-09-27:
you cannot preview and record locally at the same time