KonradIT/goprowifihack #90: HERO4 Version 5.00

IvanRech, 2017-10-01:
I updated my Hero4 with the new 5.00 firmware but it seems there is something different in the answer to the status request.
In my application I ask the status every 1 sec with:

http://10.5.5.9/gp/gpControl/status

now the camera answer one times every 12-15 and sometimes also after 45...
It seems that  the wifi is stuck :(

any idea?

KonradIT, 2017-10-01:
Why do you ping status every 1 sec? That's only needed for HERO4 Session.

IvanRech, 2017-10-01:
I Use the status to understand if the gopro is on and to show on the display when it starts to record...so 1s  seemed smart  

I tried with 2,3,4s and it seems that with the 4s it stuck few time, but I don't know If I'm missing something :(
I would like to avoid to wait 4-5s to show if it is recording :-O

KonradIT, 2017-10-01:
I didn't experience that with my Hero 4 Black with 5.00, it could take a little bit of time like maybe 2 seconds but more than 5 seems excessive. Factory reset it and see if the delay persists.

IvanRech, 2017-10-01:
Thanks! I will try also to follow the opposite approach (that I suppose is the one you are testing), I mean, I ask the status as soon as I received the previous one, or, in the worst case after 4-5s..

KonradIT, 2017-10-01:
that's a good approach, but might end up DDoSing the GoPro server

IvanRech, 2017-10-01:
I apologize for my ignorance.. :( what you mean with "but might end up DDoSing the GoPro server"?

KonradIT, 2017-12-26:
You can use an async method to avoid the delay in the responses

closed 2017-12-26 by KonradIT