KonradIT/goprowifihack #92: Hero5 Black ; Wifi stream 720

iiAnthony, 2017-10-25:
**Problem:**
Hello,
I'm unable to get a 720p stream through wifi from a GoPro Hero5 Black.  
The API you provide is great, but whatever the parameters I set, the stream received on my laptop is always 864*480.

I tried  to set quality to high , tried to set the video resolution to high values (720p, 1080p, 4k)  , the camera changes its video param, but the stream itself  keeps going with the same res.

I also tried to modify the size in ffmpeg command options (in your GoProCamera.py" script: this succeeds in showing a "bigger" image , but do not access a better resolution.

I use your "goProKeepAlive.py" script, and connects to the local address to show the stream through opencv.
"""
gopro = GoProCamera.GoPro()
gopro.video_settings(res="4k")
sleep(5)  # To ensure the switch has happened on the camera...but does not seems to influence anything
gopro.stream("udp://127.0.0.1:10000",quality = "high")
"""

and 
"""
cap = cv2.VideoCapture("udp://127.0.0.1/10000, cv2.CAP_FFMPEG)
while(True):
    m, frame = cap.read()
    cv2.imshow("GOPCV",frame)

"""
I saw (both from your github page and from gopro website infos) that it should normally be possible to provide a 720p stream but I'm unable to get it. Is there a trick I missed somewhere ?

Thanks for your help.
Regards,

KonradIT, 2017-10-25:
You get a 720p stream, but the quality is worse than 480p of course since is a bottle neck.

iiAnthony, 2017-10-26:
Ok, the issue was not a real one, updating the firmware to the current version HD5.02.01.60.00 provides the expected commands and the produced stream is actually a 720p.
Thanks a lot

closed 2017-11-08 by KonradIT

ardianumam, 2019-06-11:
Hi @iiAnthony @KonradIT ,
My firmware is HD5.02.02.60.00, but no matter what number I set in `gopro.video_settings(res="4k")`, I still always get 864x480 resolution. Any idea? Thanks.

KonradIT, 2019-06-11:
That changes camera resolution. You need 

gpCam.gpControlSet(constants.Stream.WINDOW_SIZE, constants.Stream.WindowSize.W720)

ardianumam, 2019-06-11:
Many thanks @KonradIT . It works. Let me confirm once more: (i) `gpCam.video_settings(res="value")` is to change resolution in the camera and (ii) `gpCam.gpControlSet(constants.Stream.WINDOW_SIZE, constants.Stream.WindowSize.R720)` is to change stream resolution. Is it correct? In case we have different resolution between (i) and (ii), the original image resolution is (i), then in (ii), it only resizes the frame from (i) to the size of (ii)?

KonradIT, 2019-06-11:
i) yes

2) yes

>In case we have different resolution between (i) and (ii), the original image resolution is (i), then in (ii), it only resizes the frame from (i) to the size of (ii)?

Yes