KonradIT/goprowifihack #93: Basic USB Controller Tutorial?

Quakeman0, 2017-11-03:
I am having trouble with power consumption using wireless on the Hero 5 black. Would there be any way to script up a microcontroller to give it commands through the usb? I am kind of hoping that the wifi protocol is pretty much the same, minus the http get parts? If not, any guidance is greatly appreciated :)

KonradIT, 2017-11-08:
Buy a Karma grip and a logic analyzer and leak the signals :)

Quakeman0, 2017-11-09:
I might just do that actually. This is for work so they might pony up the
cash for the analyzer and karma grip.

KonradIT, 2017-11-09:
if its for work I'd recommend joining the GoPro Developer Labs: http://developer.gopro.com

Quakeman0, 2017-11-09:
I'd like to have freedom to share the stuff I found. Also I was going to
sign up to the gopro dev but their page wouldn't let me continue past
entering my contact info and work info... so screw em.

KonradIT, 2017-11-09:
Wow, OK. [Hypoxic](https://gethypoxic.com/blogs/technical/gopro-hero5-interfaces) seems to have reverse engineered some of the protocols so you can try some of the stuff. Please post your findings! I don't have much experience in the MechEng field other than playing with oscilloscopes but I hope you can extract the signals from the USB transmission.

dengyunjie, 2018-01-23:
Our team is developing a USB control GoPro that can be discussed together。

KonradIT, 2018-01-23:
@dengyunjie nice! anything more that can be shared?

dengyunjie, 2018-01-23:
We already know that he is controlled by the USB_CDC protocol, the specific implementation way we haven't done, will do USB control after the WiFi control。
After doing it, I'll share some of our progress and results。

Calcyfer, 2018-02-20:
@dengyunjie Any news so far?

Quakeman0, 2018-02-20:
I'd be interested in just a plain old usb interface since that saves a lot
of power for my timelapse application. If you have some kind of
documentation on this and usage I'd greatly appreciate that.

Quake

Quakeman0, 2018-02-22:
Sorry to be a bother but, any word on that documentation for USB control?

neuhaus, 2020-03-29:
I‘m looking for USB reverse engineering as well. Is there a better olace for discussion than this github issue?
In particular i‘m wondering if video can be streamed via USB C with a GoPro Hero5 Session.