KonradIT/goprowifihack #95: GoPro Hero5 - Connect to Wifi Network Section - API Clarification

carlosanunes, 2017-11-11:
**Problem:**

Reading this section - https://github.com/KonradIT/goprowifihack/blob/master/HERO5/HERO5-Commands.md#connecting-to-a-wifi-network - of the documentation on the Hero 5, I get the idea that the camera can actually connect to an AP. I am I reading this right ? I tried to use it, and although I was able to scan for wireless networks, I do not seem to have been able to connect to one.

Either way just asking for a bit of clarification on the section. :)

KonradIT, 2017-12-26:
You can connect but the psk needs to be 1 for WPA2PSK.

closed 2017-12-26 by KonradIT

marktgodfrey, 2018-01-18:
@KonradIT  can you elaborate?  I'm trying connect to a network with WPA2PSK security.  what do you mean "psk needs to be 1".  would be awesome to get this working.  thanks for everything!!

demiantres, 2018-11-28:
Has anyone got this working? I can list the networks but "http://10.5.5.9/gp/gpControl/command/wireless/ssid/select?ssid=myrouterssid&auth_type=1&pw=password" does not work and returns an empty JSON string.

Also "http://10.5.5.9/gp/gpControl/command/wireless/ssid/save?ssid=myrouterssid&auth_type=1&pw=password" returns an empty JSON string.