KonradIT/goprowifihack #101: Bluetooth API for waking HERO5 from wifi-sleep

ratnick, 2018-01-03:
Problem:
HERO5 wifi connection times out when cam not recording. Need guidance for how to re-awake the cam using Bluetooth.

Description:
We are using GoPro professionally in the field doing interviews. To be able to sync in-field notes with the video, we have developed a Java program which runs on a PC and a Mac. It is based on 
https://github.com/DomDomHaas/gopro-wifi-API and works fine on HERO 3.

But after upgrading the Java API to comply with the HTTP API for HERO5, I discovered that HERO5 has a time-out on the wifi connection, when not recording. This happens even if I constantly poll the camera status.
According to https://community.gopro.com/t5/Cameras/Hero5-Session-Wifi-vs-Bluetooth/td-p/30201 this is intended behaviour to save battery. It also states that Bluetooth is the mechanism to wake up the cam again.

But I am lacking a prescription for how to wake up the camera using Bluetooth. The API description only states how to do the pairing, but not how to actually re-awake the camera programatically.

I'm sure I'm not the only one facing this kind of issue. Any help would be much appreciated.
Once the application is finished, i will of course share it.

* GoPro Camera(s): HERO5
* Firmware Version: latest (dec 2017)
* Steps to reproduce: any scenario not recording.
* Happens every time? yes

mtbsteve, 2018-01-03:
Same problem here. The Hero5B wifi is available for about 5 minutes after the camera has been turned off and WOL works fine during this time to turn on the gopro again, but after this period wifi is apparently turned off. 
I would love to see a solution which extends the wifi standby time rather than to use BT. 
Any solutions appreciated.

ratnick, 2018-01-08:
@mtbsteve: Does WOL extend the timeout period? In which case a workaround could be to send a WOL every minute or so. Clumpsy, I know

ratnick, 2018-01-08:
Yep - so would I. I have tried several ways to extend the wifi-connection time. I have tried (without success) every minute to:
- send WOL
- send http status request
- start/stop a recording (via http)
- disable + enable wifi connection (from the PC)

But no matter what I do, the gopro terminates the wifi session after 10 or so minutes. When this happens, it's like cutting the branch you sit on, and there is no way of getting wifi back, except using the re-connect UI on the camera itself. 

I wonder if there is *any* way around using bluetooth?

mtbsteve, 2018-01-08:
I tested with my iPhone and the Gopro app connected. 
1. Wifi on, Bluetooth on: the GP5B stays connected forever 
2. Wifi on, Bluetooth off: after 10 min sharp, the Gopro awakens, switches to multishot, takes automatically one multishot, and then continues to stay on
3. iPhone connected to Gopro wifi, app is closed: wifi connection shuts down exactly after 10 min. While the gopro is still turned on and gopro settings still shows wifi enabled, the gopro access point no longer visible  in the iPhone settings. Connection can be reestablished only with bluetooth turned on via the gopro app. 

So besides a parallel bluetooth connection, there seems to be a way to keep wifi awake by a mode change plus shutter release within the 10 minutes. 
Further testing needed....

ratnick, 2018-01-28:
Thanks. It inspired me, so what I do now is:
1. Connect the PC to the cameras hotspot, and then my PC application to the camera.  
2. Right after (or before, doesn't really matter), connect to the camera with the GoPro app. Stay on that connection for at least 10 minutes. It seems like the app only needs to send the "keep alive" to the camera once, because you can close the app hereafter.

The trick is that the PC can operate the camera using HTTP commands while the cam is the same time connected to the phone (which keeps the cam's wifi alive).

It's simple...but really a messy solution. I simply wonder why GoPro has not implemented a "disable wifi timeout" setting for users who has the camera connected to an external power source.

It seems to work though. Thanks to mtbsteve's inspiration :-)

mtbsteve, 2018-01-28:
Well I don’t believe that this is a viable solution but rather a bug in the gopro firmware.  I noticed that after connecting with the gopro app under some circumstances the gopro wifi did not switch off at all even after disabling the wifi in the camera. The only proper solution is with a combination of wifi and a BT connection in parallel.

KonradIT, 2018-01-28:
You can keep alive the gopro from my [api](http://github.com/konradit/gopro-py-api).

```
gopro.KeepAlive()
```

ratnick, 2018-01-29:
@mtbsteve: I agree - this is not a real solution. It's a "hack" or work-around.
@KonradIT: Interesting. My PC program is in Java, so I would need to port your KeepAlive. 

Is this the relevant line?
sock.sendto("_GPHD_:0:0:2:0.000000\n".encode(), (self.ip_addr, 8554))

And does it use a bluetooth connection?
(sorry for my ignorance here, but I'm really not familiar with sockets and the syntax used here.

thanks :-)

KonradIT, 2018-01-29:
yes that's the line.

ratnick, 2018-01-29:
OK, good.
Can you elaborate what it does? 
Is there any documentation behind this? 
Do you use bluetooth or wifi?

KonradIT, 2018-01-29:
wifi. look for keep alive on Wikipedia

mtbsteve, 2018-01-29:
Excellent-thanks @KonradIT 
Do I need to constantly send this message to keep it alive? What is the recommended interval?

ratnick, 2018-03-07:
Hi @eried 
No - i did not try because I could not find any supporting documentation, and without that, it's just a big guess-work. So i stayed with the solution depicted in my post, i.e. letting a mobile phone app handle the keep-alive for the first 10 minutes. Hereafter it seems like the camera keeps itself alive.
But admittedly - it's a crappy "solution". But the best i could do with the time I had for this.

eried, 2018-03-07:
@ratnick but do not you have issues after a while? like a couple hours? I am doing the same but for some reason my gopro stops answering my requests (after a day or so) until I connect via the phone again.

KonradIT, 2018-03-07:
@eried what camera are you using? Bluetooth might be needed to ping the GoPr9

eried, 2018-03-07:
@KonradIT Hero Session 5

You mean, on some cameras there is no way to keep them alive without bluetooth? I only need to shoot a photo every hour and retrieve the photo.

KonradIT, 2018-03-07:
Yes @eried , you need to first send a Bluetooth message to wake it up.

eried, 2018-03-07:
But after the first message, can I keep it up with the UDP message you are sending in your wrapper? @KonradIT or I will need the phone all the time to wake it?

KonradIT, 2018-03-07:
Also try to have a Magic Packet sent every few seconds like in here: http://github.com/konradit/gopro-py-api

ratnick, 2018-03-07:
@eried it seems to only need 10 minutes with the phone app. Aftern then, you can close the app and the gopro stays on. Tested on GoPro 5.

KonradIT, 2018-03-08:
@eried @ratnick The wake on lan issue affects only the H5 session or H5 Black as well? I will try to "sniff" the Bluetooth wake message.

eried, 2018-03-08:
@KonradIT I only have the H5 session, and yes it sleeps after a while. However,

@ratnick you are totally right, thanks! after leaving my phone 15 mins with the app opened I have been running my script connected to the GoPro for 1 day without any issues!

KonradIT, 2018-03-23:
Hey guys, I can wake up the camera using [this app](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp), we can even get fw info.

eried, 2018-03-23:
Just connecting to the camera?

KonradIT, 2018-03-23:
Yes. My laptop's BT controller does not pair with the GoPro for some reason, but theoretically using bluetoothctl trust > pair > connect should work.

KonradIT, 2018-03-23:
https://streamable.com/tjb3t

memento87, 2018-04-30:
Here's my solution (GOPRO HERO 5):
Manually pair the camera with your laptop's (or raspberry pi, or whatever device) bluetooth adapter.
To do this, scan for the camera in your computer's bluetooth pairing menu, then open the go pro's connect menu > Connect New Device > GoPro App.
That's all. Now all you need to do is connect to your gopro from bluetooth and the camera automatically wakes up and brings the wifi up with it.

I wrote an app in java (using the tinyb bluetooth library) and here's my code:

`
BluetoothManager manager = BluetoothManager.getBluetoothManager();
manager.startNearbyDiscovery(new BluetoothDeviceDiscoveryListener()
{
	@Override
	public void bluetoothDeviceDiscovered(BluetoothDevice device)
	{
		System.out.println(device.getAlias());
		device.setTrusted(true);
		device.connect();
		device.disconnect();
	}
}, 10000, false);
`
I understand that ble can be used to perform more stuff on the gopro such as trigger the shutter and change the settings, but I find this redundant. All I needed was the camera to wake up and the wifi to go back on, then I can do all those stuff using the http apis you guys have listed here.

Hope this helps others as I wasted a day trying to decompile the gopro apk, and sniffing bluetooth packets from btsnoop_hci to no avail... it turned out to be much easier than that :)

krystof-k, 2018-05-10:
Hey, anyone figured out how to keep the HERO5 alive & connected to WiFi? Thanks!

krystof-k, 2018-05-11:
I have some progress. I bought Raspberry Pi with BLE yesterday and I'm able to wake up the GoPro over Bluetooth. However still not figured out how to turn on the WiFi (but at least there is a way to turn GoPro off over WiFi and then turn it on over BLE).

KonradIT, 2018-05-11:
@krystof-k Which RPI did you buy? Mi Pi Zero stopped working all of a sudden.

eried, 2018-05-11:
How do you turn it on? Just linking it via BT does not keep my GoPro on :/

krystof-k, 2018-05-12:
I made it! Able to turn on the WiFi (and much more) over BLE. I’ll try to write some tutorial and create PR.

krystof-k, 2018-05-13:
So, [here](https://github.com/KonradIT/goprowifihack/pull/114 it is!

Someone please try so that I'm sure I'm not the only one :))

KonradIT, 2018-05-13:
Wow that's cool! Going to test now with a Macbook to see if it works, my laptop cannot pair with the Hero5 BT

krystof-k, 2018-05-24:
Hmm, just found out that when GoPro loses power, it doesn't turn it automatically on. In that case, the only option is to manually turn it on pressing the button. That's a problem :(

KonradIT, 2018-05-25:
>loses power

What do you mean by loses power? Battery empty?

krystof-k, 2018-05-25:
When running without the battery. Probably the same case would be when freezing/overheating.

krystof-k, 2018-06-28:
@me89 Thanks! :) Well, not into a detail, you can learn more [here](https://gethypoxic.com/blogs/technical/gopro-hero5-interfaces), but generally `03170101` is a HEX byte sequence which you can either assemble from the [blogpost](https://gethypoxic.com/blogs/technical/gopro-hero5-interfaces) or sniff it by Android app when using GoPro app. But I guess the option to turn on the WiFi should be enough because after that you can controll everything over WiFi.

It can work as a keep-alive, kind of, however I don't have tested if the camera still doesn't turn off the WiFi after 10 minutes. But you can call it always before using WiFi and test whether you're connected to the camera, so in case it does you'll always turn the WiFi on before proceeding with the requests (that's my approach).

me89, 2018-07-01:
I tested it, it doesn't turn off the camera after 10 minutes. I send the "Keep-Alive" signal every minute. With it, its possible to have a reliable remote control for the GoPro5. I integrated it in nodejs as child_process.

sebasobotka, 2018-07-04:
@me89 how do you send this "keep-alive" signal? It's over BT? Do you send those 4 bytes 03170101 (i assumed the first byte is the length of the message). Is it possible to pair gopro camera with BT on PC?

sebasobotka, 2018-07-05:
@me89 thanks for your reply. Also thanks to @KonradIT. Good work !
Based on your work I am developing a simple app in openframeworks (C++) and I had the same issues with turning the wifi off also. I am working with gopro 5 and 6. 

I need a camera to record a short movie by request.
I check gpMediaList every 1sec via http and start/stop recording a movie every 8min (less than 10) when it's in idle. This works fine and keeps the camera alive but it records movies which I don't need. 

Yesterday I found another way. I send a request via http to set a camera to the particular mode every 1-8 min (less than 10)
http://10.5.5.9/gp/gpControl/command/sub_mode?mode=0&sub_mode=0
I set the camera to the Video (VIDEO) mode in my case which stays in the same mode all the time. This works fine and I keeps camera alive with wifi (without using BT). Tested with gopro6.

KonradIT, 2018-07-09:
Using the [commands list](https://gethypoxic.com/blogs/technical/gopro-hero5-interfaces) here are some copy-paste ready gatttool commands:

````
RECORD STOP: char-write-req2f 03010100
RECORD START: char-write-req2f 03010101

VIDEO: char-write-req2f 03020100
PHOTO: char-write-req2f 03020101
MULTISHOT: char-write-req 2f 03020102

SINGLE VIDEO: char-write-req 2f 050301000100
````

Layout for command is [cmd len][byte1][byte2][etc...]

rcywongaa, 2018-07-09:
The magic `_GPHD_:0:0:2:0.000000\n` packet didn't work for me (maybe I did something wrong) 
but starting streaming via [`http://10.5.5.9/gp/gpControl/execute?p1=gpStream&a1=proto_v2&c1=restart`](https://github.com/KonradIT/GoProStream/blob/master/GoProStream.py) successfully prevented my gopro from turning off wifi

KonradIT, 2018-07-13:
http://github.com/konradit/gopro-ble-py

CLI-like BLE controller for Hero 5 using Python. Would appreciate some feedback.

krystof-k, 2018-07-13:
Great work @KonradIT!

npaisnel, 2018-08-06:
I understand my question is not directly related to this articular issue , but I can see how it will affect my current project.

Seem to be very little information and projects about related to connecting to and controlling a Hero 5 from anything other than the OEM app.

I am looking at creating an Arduino based motion detector to take a photo or video when motion is detected.  

A plug in trigger for the Hero5 would be idea, but since such a thing does nto seem to exist it has to be BT/WiFi control.

Just posting here to see if any one has any handy threads that may help me in this..links to code or the protocol used , what signals need to be sent to trigger the camera, keep it awake etc.

This is all just an idea yet, been brewing for a while, but now with 6 months laid up with a broken leg ahead of me, time to do some research to see if this is achievable

krystof-k, 2018-08-15:
Hmm, unfortunately, it's not very reliable. For the second time, GoPro stopped responding to Bluetooth commands :(

KonradIT, 2018-08-15:
@distancerunner Send a Magic Packet every 2.5 seconds as indicated here: https://github.com/KonradIT/GoProStream/blob/master/GoProStream.py#L108

@npaisnel as @krystof-k said, the Bluetooth connection is not very reliable. You might have a chance to get this work by keeping the camera alive (see above), with a constant power supply and a fan. Unless you need high quality stills from the GoPro opt for a trail camera.

npaisnel, 2018-08-16:
Ha ! 👍 A trail cam was the start of this idea, but mum decieded to go for the better image quality of the GoPro, meaning she could use it for her other stuff like cycling , sea kayaking, skiing etc.

Then I got the job of finding the way to make the GoPro movement activated.   Just amazed that with all the accessories for the GoPro out there, there is not a single off the shelf item except for the Cam_Do system that will end up costing more than the Go-Pro

habakuk39, 2019-05-01:
@konradIT: do you have an installable package for your BLE-Controller for python (http://github.com/konradit/gopro-ble-py)?

so that I can install it like I installed your great api:

"pip install goprocam"

KonradIT, 2019-05-02:
@habakuk39 no.

_Sent from my Mi A1 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

habakuk39, 2019-05-02:
@konradIT: problem with wifi on/off command over BLE
hero5black connected to linux: unfortunately it happens every time:

first everythings works fine with: bluetoothctl --> trusted, paired, connected --> then disconnect again;

then sudo gatttool.... --> connect; ok, everything fine, I can even do the locate commands:
char-write-req 2f 03160101 and 
char-write-req 2f 03160100

but when I try to manipulate wifi
char-write-req 2f 03170101 or
char-write-req 2f 03170100

I always get the "Invalid file descriptor" Error:
(gatttool:10112): GLib-WARNING **: 08:07:44:201: Invalif´d file descriptor

then:
- gatttool keeps on disconnecting
- and even bluetoothctl, still scanning the device, cannot re-pair or reconnect the device anymore!

so by:
char-write-req 2f 03170101    to my gopro hero5Black
the whole device-pairing gets corrupted! Why is that.

the only way to fix this is to reset all connections on the gopro to get a new Bluetooth address, on bluetoothctl re-scan, re-pair, ... and start all over.

Any fix for that issue?

KonradIT, 2019-05-02:
Weird I'll try to reproduce it this night with my 5 black.

habakuk39, 2019-05-02:
> Weird I'll try to reproduce it this night with my 5 black.

thank you in advance, will appreciate any input....

habakuk39, 2019-05-03:
@KonradIT: anything new from your side about my issue?

KonradIT, 2019-05-03:
@habakuk39 

>my 5 black is NOT connected via wifi to the bluetooth paired linux! (odroid C2 - no wifi yet...;-))

I haven't been able to pair my HERO5 Black via BT unless the PC was connected via WiFi to it beforehand.

habakuk39, 2019-05-03:
@konradIT: ok, but in case of wifi sleep mode of hero5 the pc isn't connected via wifi anyway, right? but still paired via BT. so why is 
char-write-req 2f 03170101
then going to destroy blue tooth pairing?

or am I understanding something wrong?

KonradIT, 2019-05-04:
GoPro never leaves pairing mode because you have to use WiFi to confirm pairing.

habakuk39, 2019-05-04:
ok, so I have to wait for my wifi modules (Hardkernel) and then I hope everything will work out fine.

hopefully. I will be back and let you know....

kind regards

habakuk39, 2019-05-14:
> Hmm, just found out that when GoPro loses power, it doesn't turn it automatically on. In that case, the only option is to manually turn it on by pressing the button. That's a problem :(

@KonradIT
@krystof-k 

You are right. When removing/moving back in the battery of the gopro you cannot wake up the gopro anymore over ble, nor does it of course reastblish a wifi connection.
You have to press the button on the camera to switch it on!

Same happens of course, running the cam ob wired power supply without battery in it, so that you switch it of by removing the power cable. When you plug the power cable back in  agian, the back led blinks one times red, but right afterwards there is no ble/wifi connection coming up, either

To use it e.g. as a dash cam for me it's superimportant to get the camera reconnected completely unattended.

Any Ideas/news on that issue?

[16 more replies]