KonradIT/goprowifihack #102: unable to use the wifi commands

aster94, 2018-01-13:
**Details:**

* GoPro Camera(s): hero 3+ silver
* Firmware Version: 3.10.03.02
wifi version: 4.0.36.1
* Steps to reproduce: using a esp01
* Happens every time? [Y/N]:y
my password is password

**Problem:**
i tried to control my gopro with esp8266 (esp01) using a serial-ttl converter
i sent:
```
AT+CWMODE=1
AT+CWJAP="Saiken-GoPro","password" 
AT+CIPMUX=0
AT+CIPSTART="TCP","10.5.5.9",8080
```
until here all works ok, now i should send the request
```
AT+CIPSEND=53
http://10.5.5.9/bacpac/SH?t=password&p=%01 HTTP/1.0
```
nothing happens
i tried:
1 to turn it on or off, control the shutter
2 change bacpac to camera 
3 put a "GET " before the request
4 change "HTTP/1.0" to "HTTP/1.1"
5 don't put the final " HTTP/1.0"

nothing worked, is i possible that the last version ( 2 september 2015 as stated here: https://gopro.com/update/hero3_plus) of the wifi firmware changed the commands?

is any simple way to "see" which commands use the gopro app in my phone?
from the gopro app on my android phone it works ok

KonradIT, 2018-01-13:
is your password "password"? then is should work. try to ping 10.5.5.9

aster94, 2018-01-13:
konrad, i have to admit that my knowledge about "internet" are not so good
i pinged the addres of the gopro, i run on ubuntu 16.04
```
ping -c1 10.5.5.9
PING 10.5.5.9 (10.5.5.9) 56(84) bytes of data.

--- 10.5.5.9 ping statistics ---
1 packets transmitted, 0 received, 100% packet loss, time 0ms
```
i also tried traceroute:
```
traceroute  10.5.5.9
traceroute to 10.5.5.9 (10.5.5.9), 30 hops max, 60 byte packets
 1  modemtim.homenet.telecomitalia.it (192.168.1.1)  0.903 ms  1.186 ms  1.379 ms
 2  * * *
 3  * * *
 4  * * *
 5  * * *
 6  * * *
 7  * * *
 8  * * *
 9  * * *
10  * * *
11  * * *
12  * * *
13  * * *
14  * * *
15  * * *
16  * * *
17  * * *^Z
[7]+  Fermato                 traceroute 10.5.5.9
```

aster94, 2018-01-17:
hola konrad,
it looks like that i was using the wrong port: i read somewhere in a forum that the gopro hero3+ uses 8080, but really to use the wifi commands you should use 80

by the way while i was investigation this i make a library in C/C++ which will support a lot of MCU: all arduino compatibles boards (+wifi module) and esp8266 boards, probably also the esp32. I plan to add more cameras in the future

you may be interested to add this to your list of 3rd party libraries since you don't have any which run on microcontrollers

aster94, 2018-01-17:
i completely forgot to put a link to the library 
https://github.com/aster94/GoProControl
and a video if you want to see it in action:
https://www.youtube.com/watch?v=PuM-ZQ2tMW0