KonradIT/goprowifihack #105: WiFi will be broken in a few minutes

dengyunjie, 2018-02-01:
**Problem:**
First  I send a magic packet (WoL) with these parameters: MAC ADDRESS OF THE CAMERA, 10.5.5.9 as IP ADDRESS, Subnet Mask 255.255.255.0, Port 9. GET the status 31 of the camera，It was found that it was 0. WiFi will be broken in a few minutes，

Describe your problem as accurately as possible, include key details that help resolve your issue.

**Details:**

* GoPro Camera(s): GOPro5
* Firmware Version:
* Steps to reproduce: After WiFi connection, in a few minutes, GoPro WiFi closes
* Happens every time? [Y/N]: Y

Narayan1986, 2018-03-21:
I have the same problem. After 10 minutes GoPro stop WiFi...
I try to send an info request every minute but problem is the same...
Someone find a solution ?
Model HERO5

KonradIT, 2018-03-23:
https://github.com/KonradIT/goprowifihack/issues/101#issuecomment-375711393