KonradIT/goprowifihack #108: hero4 on linux with 2 wifi APs streaming to v4l2loopback

Gunstick, 2018-03-25:
**Problem:**

Was missing instructions on how to set up wifi, connect to gopro and get the stream and then send that stream to the Internet (i.e. skype). I figured all that out (see below, maybe to add to a readme)

Now I'm stuck with getting ffmpeg to play nicely and dump the video into /dev/video1 (the v4l2loopback device). Suprisingly ffplay works fine, but ffmpeg not:
[mpegts @ 0x22b7fa0] Could not find codec parameters for stream 1 (Audio: aac ([15][0][0][0] / 0x000F), 0 channels, fltp): unspecified sample rate
Consider increasing the value for the 'analyzeduration' and 'probesize' options
[mpegts @ 0x22b7fa0] Could not find codec parameters for stream 2 (Unknown: none ([128][0][0][0] / 0x0080)): unknown codec
Consider increasing the value for the 'analyzeduration' and 'probesize' options

**Details:**

* GoPro Camera(s): hero4
* Firmware Version: 
ffmpeg version 2.8.11-0ubuntu0.16.04.1 Copyright (c) 2000-2017 the FFmpeg developers

* Steps to reproduce:
ffmpeg -re -i udp://10.5.5.9:8554 -map 0:v -f v4l2 /dev/video1 

* Happens every time? [Y/N]: Y

Documentation:
What I already did to make this work.
Laptop with additional USB WiFi adapter.

Switch GoPro to AP mode, if not already synched with smartphone, do so to set up SSID and password.

Plug USB-WiFi adapter in, check if it sees your GoPro, connect to that SSID
There should now be 2 networks on your laptop

$ ip address

But the route is wrong, you get 2 default routes. Just delete the one for GoPro

$ sudo ip route del default via 10.5.5.9

Check if you can reach the internet again: ping -c 1 8.8.8.8
Now you are set to tell the gopro to stream
$ wget -O /dev/null 'http://10.5.5.9/gp/gpControl/execute?p1=gpStream&a1=proto_v2&c1=restart'

Watch the stream:
$ ffplay -fflags nobuffer -f:v mpegts -probesize 8192 udp://10.5.5.9:8554

Set up v4l2loopback
sudo apt-get install v4l2loopback-dkms
sudo modprobe v4l2loopback

you now have an additional video device 
ls -l /dev/video*

Use that one to dump the stream into
ffmpeg -re -i udp://10.5.5.9:8554 -map 0:v -f v4l2 /dev/video1    # this part not working currently

You now can use /dev/video1 for skype and other video applications over the internet.

skype on linux (ubuntu): sudo snap install skype 
or in my case: sudo snap install skype --classic

KonradIT, 2018-03-25:
I was attempting this yesterday as well!

What is your log for  ffmpeg -re -i udp://10.5.5.9:8554 -map 0:v -f v4l2 /dev/video1? Mine could not stream to /dev/video0 because of the codec.

Gunstick, 2018-03-25:
Yeah similar, but I have no idea how to solve the codec issue. Someone knows stuff about ffmpeg?
```

Input #0, mpegts, from 'udp://10.5.5.9:8554':
  Duration: N/A, start: 6.120000, bitrate: N/A
  Program 1 
    Stream #0:0[0x1011]: Video: h264 (Main) ([27][0][0][0] / 0x001B), yuvj420p(pc, bt709), 320x240 [SAR 1:1 DAR 4:3], 25 fps, 25 tbr, 90k tbn, 50 tbc
    Stream #0:1[0x1100]: Audio: aac ([15][0][0][0] / 0x000F), 0 channels, fltp
    Stream #0:2[0x200]: Unknown: none ([128][0][0][0] / 0x0080)
[v4l2 @ 0xdb52a0] Unknown V4L2 pixel format equivalent for yuvj420p
Output #0, v4l2, to '/dev/video1':
  Metadata:
    encoder         : Lavf56.40.101
    Stream #0:0: Video: rawvideo (I420 / 0x30323449), yuvj420p, 320x240 [SAR 1:1 DAR 4:3], q=2-31, 200 kb/s, 25 fps, 25 tbn, 25 tbc
    Metadata:
      encoder         : Lavc56.60.100 rawvideo
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> rawvideo (native))
Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument

```

johngardenier, 2018-11-20:
The following worked for me on ubuntu 16.04 and hero5 black. The delay was around 1 second. Note that you can set the stream bitrate and resolution listed in goprowifihack/HERO*/WifiCommands.md

In one terminal run this script to keep gopro alive (this command may be slightly different for other gopro versions, have a look here for your camera: goprowifihack/HERO*/Livestreaming.md):

`while true; do wget -O /dev/null 'http://10.5.5.9/gp/gpControl/execute?p1=gpStream&a1=proto_v2&c1=restart'; sleep 5; done`

In another terminal you can then run the following to stream to /dev/video4 (setup virtual video device with sudo modprobe v4l2loopback video_nr=4):

`ffmpeg -i udp://10.5.5.9:8554 -fflags nobuffer -f:v mpegts -probesize 8192 -map 0:v -pix_fmt yuv420p -r 20 -f v4l2 /dev/video4`