KonradIT/goprowifihack #109: Live Stream GoPro Hero 4 Silver  V5.0

Trulde, 2018-03-27:
Hi,

I gonna try to get the livestream of my GoPro Hero 4 Silver. 
First I tried: 
1. cmd: ffplay udp://:8554
2. open http://10.5.5.9/gp/gpControl/execute?p1=gpStream&a1=proto_v2&c1=restart 
 => status 0

Second I tried your GoProStream.py script.
I run the Script and get a connection, but no livestream

branch HD4
HERO4 Silver
HD4.01.05.00.00
UDP target IP: 10.5.5.9
UDP target port: 8554
message: _GPHD_:0:0:2:0.000000

I was able to start recodring of the camera - so the connection is fine in my opinion.

Is it impossible to stream with the actual firmware?

* GoPro Camera(s): Hero 4 SIlver
* Firmware Version: V.05.00
* Steps to reproduce: 
* Happens every time? [Y/N]: Y

KonradIT, 2018-03-27:
Change udp://:8554 to  udp://10.5.5.100:8554 in line 94

Trulde, 2018-03-27:
Thank you for quick response. It is working fine now.

closed 2018-03-27 by Trulde