KonradIT/goprowifihack #110: Live Video Resolution

eviling, 2018-03-27:
**Problem:**

Hello,

I am using the ffmpegless-preview.py to do a live video with a GoPro Heroe5. It connects perfect, but the frame using cv2.VideoCapture() is (864,480), but the real resolution of the camera is 1080.

The problem is that I want to get a frame with the original resolution (1080p) using OpenCV.

camera overview
current mode: Video
current submode: Video
current video resolution: 1080p
current video framerate: 120
pictures taken: 35
videos taken:  31
camera model: HERO5 Black
firmware version: HD5.02.02.00.00

If a check the width and height of OpenCV, I get the 864 an 480.

`cap = cv2.VideoCapture("udp://10.5.5.9:8554")
while True:
    nmat, frame = cap.read()
    width = cap.get(3)  # float
    print(width)
    height = cap.get(4)
    print(height)
    cv2.imshow("GoPro OpenCV", frame)
  `

KonradIT, 2018-03-29:
The stream resolution is limited to 720p.

KonradIT, 2018-05-28:
@chrisrb32 I did not test that, but you can test it out yourself