KonradIT/goprowifihack #111: Can connect and control Go Pro - But can't take photo/video

gngriffiths, 2018-04-10:
**Problem:**

Currently using a GoPro4, but have been testing with the Hero also.

I can connect to the GoPro, changing settings and pull down files, but I can't take a photo or record a video. Did I miss something in the documentation?

Thank you for a great hack =)

d3mac123, 2018-04-10:
I just tested it here and it worked. However, I had to "force" the video mode on first (not sure if this is necessary thou). First line force video mode. The second line activates the "recording" button:
http://10.5.5.9/gp/gpControl/setting/10/1
http://10.5.5.9/gp/gpControl/command/shutter?p=1

gngriffiths, 2018-04-11:
Really helpful thank you. I missed the Shutter control setting.

closed 2018-04-11 by gngriffiths