KonradIT/goprowifihack #116: Change ip and port?

julianraheema, 2018-06-05:
**Problem:**
When I connect two gopros over two wifi on Linux, the cameras are connected but on the browser 10.5.5.9/gp/gpControl it talks only one gopro and response to the browser command such as sleep. But not the second gopro. In theory if I have different ip and port for the second gopro, I might be able to access them both using the browser then I would be able to stream both using python script to live stream both using udp on openCV since both cameras have different ip and port.

**Details:**

* GoPro Camera(s): Hereo 5 black 
How to change the gopro IP: 10.5.5.9 and Port: 8554?

afogal, 2018-06-13:
I'm actually having the same issue, I'm looking through the HERO4 firmware to see if it is going to be an easy change

afogal, 2018-06-21:
Okay, here is how to do this:

1. Use some kind of custom firmware so you can send commands to the linux part of the GoPRo
2. Issue the following commands:
```
ip addr add 10.6.5.9/24 dev ap0 brd 10.6.5.255
ip route add 10.6.5.0/24 dev ap0 src 10.6.5.9
ip addr del 10.5.5.9/24 dev ap0
ip route del 10.5.5.0/24 dev ap0
```
Tested on GoPro HERO4 Silver. You might need to find the interface yourself.

KonradIT, 2018-06-21:
@afogal what firmware is that? I know Alti-force lets you execute override.sh

afogal, 2018-06-21:
Yes. that is the firmware I am using, and the commands are an excerpt of my override.sh. I do not know if a similar thing is available for other GoPro cameras, but the same commands should work as long as the interfaces stay the same.

julianraheema, 2018-06-25:
@afogal thank you for your comment. I will try it.