KonradIT/goprowifihack #117: Get MAC address of GoPro camera

huangwaylon, 2018-06-14:
**Problem:**
When and how can I retrieve the MAC address of the GoPro camera?

I want to turn on the device by sending the wol message. 

According to the general WIFI commands page I need to:

> To power ON a HERO4 Black/Silver camera, use WoL with these parameters: MAC ADDRESS OF THE CAMERA, 10.5.5.9 as IP ADDRESS, Subnet Mask 255.255.255.0, Port 9. For the Hero4 Session, this magic packet has to be sent when your project wants to operate with the camera.

Is this information sent back from the camera in some http response?

**Details:**

* GoPro Camera(s): Hero 5 Black
* Firmware Version: 2.2.0

huangwaylon, 2018-06-14:
Never mind, got it with /info!

closed 2018-06-14 by huangwaylon

iceman-k, 2018-08-13:
@huangwaylon hi, I have a same problem, could you tell me where /info is? Thanks

iceman-k, 2018-08-13:
Sorry @huangwaylon, I've just got it.

http://10.5.5.9/gp/gpControl/info

This url gave me mac adress of GoPro. Thanks