KonradIT/goprowifihack #120: Pairing a Raspberry Pi

Chase-T, 2018-07-27:
Hi Konrad,

Thanks for all the great work you have done on this so far. I can't imagine the patience and dedication required to figure all this out.

I have a Raspberry Pi B+3 that i am trying to pair with my new Hero Session 5.

I have three problems - Wifi, Bluetooth and the python API.

Wifi
I can connect to the wifi.
I can't ping 10.5.5.9.
I can't control the GP through the web page.

Bluetooth
I can pair the Bluetooth, but it wont stay connected. 
My terminal output looks something like this...
pi@raspberrypi:~ $ sudo bluetoothctl
[NEW] Controller B8:27:EB:DE:E4:82 raspberrypi [default]
[NEW] Device FB:FB:E2:CD:95:D6 GoPro 3809
[CHG] Device FB:FB:E2:CD:95:D6 Connected: no
[CHG] Device FB:FB:E2:CD:95:D6 Connected: yes
[CHG] Device FB:FB:E2:CD:95:D6 Connected: no
[CHG] Device FB:FB:E2:CD:95:D6 Connected: yes
[CHG] Device FB:FB:E2:CD:95:D6 Connected: no

Python API
Your poprocam package seems to install Ok, but when i try to use it I get an error message.

pi@raspberrypi:~ $ pip install goprocam
Requirement already satisfied (use --upgrade to upgrade): goprocam in /usr/local/lib/python2.7/dist-packages
Cleaning up...
pi@raspberrypi:~ $ python

Python 2.7.9 (default, Sep 17 2016, 20:26:04) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from goprocam import GoProCamera
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/goprocam/__init__.py", line 1, in <module>
    from goprocam import GoProCamera
  File "/usr/local/lib/python2.7/dist-packages/goprocam/GoProCamera.py", line 3, in <module>
    import urllib.request
ImportError: No module named request
>>> 

Am I missing something?
Is there a step-by-step guild to pairing under linux?
Any help you could provide would be greatly appreciated.

Thanks,
Chase.

KonradIT, 2018-07-31:
Hey, you need to use Python 3.5 or later.

Chase-T, 2018-08-01:
Cool.
I will give it another go this weekend.
Thanks,
Chase.

Chase-T, 2018-08-05:
Hi Konrad,

I have upgraded to Python3.5.2, but still get an error when trying import.

goprocam shows up on the pip list.

pi@raspberrypi:~ $ sudo pip3 install goprocam
Collecting goprocam
  Downloading https://files.pythonhosted.org/packages/ee/41/bd1be1801fc715eb2b97466c283ceec158ad39dd88cedfa5a688de1a0e9f/goprocam-2.0.6.tar.gz
Building wheels for collected packages: goprocam
  Running setup.py bdist_wheel for goprocam ... done
  Stored in directory: /root/.cache/pip/wheels/4a/77/3c/ef9738cb638fb9d9981e983a898d9cbc28a4824e651514e6d7
Successfully built goprocam
Installing collected packages: goprocam
Successfully installed goprocam-2.0.6

pi@raspberrypi:~ $ pip3 list
...
cryptography (0.6.1)
Flask (0.10.1)
goprocam (2.0.6)
gpiozero (1.4.0)
...

pi@raspberrypi:~ $ python3.5
Python 3.5.2 (default, Aug  5 2018, 10:19:40) 
[GCC 4.9.2] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from goprocam import GoProCamera
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named 'goprocam'
>>> 

Any idea what i am doing wrong?

Thanks,
Chase