KonradIT/goprowifihack #122: Hero5 Black-Video Conversion not working

distancerunner, 2018-07-31:
**Problem:**
I recorded a 14s clip: GOPR0108.MP4
after that, I tried to convert/cut the clip with:
http://10.5.5.9/gp/gpControl/command/transcode/request?source=DCIM/100GOPRO/GOPR0108.MP4&res=0&fps_divisor=0&in_ms=500&out_ms=2000

The Response is:
{
    "version": "5.00",
    "path_info": "command/transcode/request",
    "query_string": "source=DCIM/100GOPRO/GOPR0111.MP4&res=0&fps_divisor=0&in_ms=500&out_ms=2000",
    "error_code": -3,
    "error_msg": "400 Bad Request\r\n",
    "function": "gpcontrol_cgi_handler",
    "line": 170
}

with the python script I tried:
gp5.getClip(100GOPRO/GOPR0108.MP4, constants.Clip.R1080p, constants.Clip.FPS_NORMAL, 0, 2000)
but here is the answer:
"
    gp5.getClip(100GOPRO/GOPR0108.MP4, constants.Clip.R1080p, constants.Clip.FPS_NORMAL, 0, 2000)
                       ^
SyntaxError: invalid syntax
"
**Details:**
Does In-Camera_Edit working on GoPro 5?

* GoPro Camera(s): Hero 5 Black
* Firmware Version: 2.60
* Steps to reproduce: 
* Happens every time? [Y/N]:

KonradIT, 2018-07-31:
Oh hey :)

http://10.5.5.9/gp/gpControl/command/transcode/video_to_video?source=DCIM/101GOPRO/GOPR1452.MP4&res=0&fps_divisor=0&in_ms=500&out_ms=2000

Now its **video_to_video** , gopro-py-api and docs will be updated.

KonradIT, 2018-07-31:
status can be checked with command/transcode/status

http://10.5.5.9/gp/gpControl/command/transcode/status?id=433791698348