KonradIT/goprowifihack #124: API to download media files (videos/photos) from GoPro HERO 2018 to Android phone

rahulpatidar009, 2018-08-14:
I am able to connect with GoPro HERO 2018, and can control some features from my android phone like, capturing videos and photos, live streaming etc. 

Now I want to **download media file (videos/photos) from GoPro and want to save them on my android phone**. There are some APIs for video/photo info, exif, thumbnail etc, but I am unable to find API for downloading media files (videos/photos).

Please help me to get out from trouble.
Thanks in advance.

KonradIT, 2018-08-14:
What have you tried before opening this issue? I can't do your job for you.

rahulpatidar009, 2018-08-14:
I have tried following APIs for downloading media files-
http://10.5.5.9:8080/gp/gpMediaMetadata?p=100GOPRO/GOPR0009.MP4 // To get thumbnail of a video
http://10.5.5.9:8080/videos/DCIM/100GOPRO/GOPR0008.JPG // For downloading specific image
http://10.5.5.9:8080/videos/DCIM/100GOPRO/GOPR0009.MP4 // For Downloading specific video

All the above APIs returning with status OK, but the content I am receiving from these APIs is not converting into Video/Photo.

If you want code which I am using for conversion, then I can also provide you that code.

KonradIT, 2018-08-14:
>http://10.5.5.9:8080/videos/DCIM/100GOPRO/GOPR0009.MP4 

That *is* a video. Try downloading it in your browser, it will play it back (or in VLC)

rahulpatidar009, 2018-08-14:
Thanks for your quick response.
Yes, I am able to play it on my browser as you have suggested.
But I want to download it on my local SD card of android phone and then play from there only.

KonradIT, 2018-08-14:
Well, how do you plan to do it? This is not the place to discuss it, research online and figure it out.

closed 2018-08-14 by KonradIT