KonradIT/goprowifihack #131: Bluetooth pairing Hero7 on Raspberry Pi fails

kmpm, 2018-11-06:
**Problem:**
I am trying to pair a Hero 7 black with my raspberry pi following the guide here and it fails with mostly
`Failed to pair: org.bluez.Error.AuthenticationCanceled` but I have seen `Failed to pair: org.bluez.Error.Failed`.

**Details:**
* /etc/issue: Raspbian GNU/Linux 9
* Raspberry Pi: https://www.raspberrypi.org/products/raspberry-pi-3-model-b-plus/
* GoPro Camera(s): Gopro Hero 7
* Firmware Version: HD7.01.01.51.00

## steps to reproduce
### Camera
* Start camera
* reset connections
* `Preferences > Connections > Connect Device > GoPro App` or  
  `Preferences > Connections > Connect Device > Bluetooth Device`

### Raspberry PI
```bash
sudo bluetoothctl
[NEW] Controller B8:27:EB:73:XX:XX raspberry [default]
[NEW] Device F9:30:42:7F:XX:XX GoPro 0022
[bluetooth]# trust F9:30:42:7F:XX:XX
Changing F9:30:42:7F:XX:XX trust succeeded
[bluetooth]# pair F9:30:42:7F:XX:XX
Attempting to pair with F9:30:42:7F:XX:XX
[bluetooth]#
```

Here it takes some time, like a minute, without any feedback at all. Then I get...

```bash
Failed to pair: org.bluez.Error.AuthenticationCanceled
[bluetooth]#
```

r2cool, 2019-04-04:
You need to enable the experimental flag for bluez to work correctly with the GoPro.
See https://learn.adafruit.com/install-bluez-on-the-raspberry-pi/installation#enable-bluetooth-low-energy-features-2-38

jsmith79, 2019-04-09:
I too have been having issues with pairing my GoPro hero 7 and my rpi3 b+. I enabled the experimental flag as per the above link but I still time out and get the “Failed to pair: org.bluez.Error.AuthenticationCanceled” message. Any ideas would be appreciated.

Thanks.

omnibrain, 2019-05-01:
Same problem with the HERO6

gafurmp, 2020-04-25:
Same problem with my HERO7 Black. Any hints?

KonradIT, 2020-05-16:
More documentation is coming as I test the BT connection flow on different FW versions and cameras.