KonradIT/goprowifihack #137: Bluetooth commands

demiantres, 2019-01-16:
The commands from https://gethypoxic.com/blogs/technical/gopro-hero5-interfaces and https://github.com/KonradIT/gopro-ble-py/blob/master/main.py differ (Konrad's values do work).
For instance gethypoxid:
` SHUTTER_ON(new byte[]{(byte) 1, (byte) 1, (byte) 1}),`
vs.
`bytearray(b'\x02\x01\x01')` (Konrad).

Why is this difference/how can the hypdoxid values be correctly converted?

KonradIT, 2019-01-16:
The first byte is the length of the packet. 2 (length of ==>), 1, 1

cepoon, 2020-04-14:
I believe the commands seen here are of length 1 (simple command, like power off or tag), length 3 (stuff like locator or shutter) or length 5 (sub-mode change). Here are some examples, sent via gatttool:
0104 = Power Off everything including BLE
0105 = Power Off but leave BLE on until timeout
0301 0100 = Shutter Off (command length = 3, length of 1st param = 1)
0503 0101 0102 = Night Shot Photo mode (command length = 5, length of 1st param = 1, length of 2nd param = 1)