KonradIT/goprowifihack #138: Change camera settings with Bluetooth commands

demiantres, 2019-01-16:
Is there a way to change camera settings (protune, resolution, and so on) using Bluetooth commands? Is there a way to get a list of possible BT commands?

cepoon, 2020-04-14:
If you already have a Linux BLE client paired to the GoPro, send a write request to b5f90074-aa8d-11e3-9046-0002a5d5c51b (Handle 52 or 0x34 on a GoPro 5 Black) like this using gatttool:
gatttool -t random -b ${GOPRO_MAC} --char-write-req -a 52 -n 03${SETTING}01${VALUE}

Where SETTING is a 2 digit hex representation of the setting number in the REST API, and VALUE is the 2 digit hex representation of the value of the particular setting. I have only tried this with a few setting where the value is below 256 (since VALUE can only be an 8-bit number). The "03" is the length of the total command, while "01" is the length of the VALUE. I have never tried sending something like this: 04${SETTING}02${VALUE_16bit}

KonradIT, 2020-04-14:
Neat! Didn't know that, mind doing a PR @cepoon ?