KonradIT/goprowifihack #139: GoPro 3 file deletion dangerously broken

f0k, 2019-01-28:
**Problem:**

`deleteFile()` sends a `DA` command to the GoPro:
https://github.com/KonradIT/gopro-py-api/blob/6a25cc8/goprocam/GoProCamera.py#L350
Furthermore, `sendCamera()` always preceeds the option with a `%` sign:
https://github.com/KonradIT/gopro-py-api/blob/6a25cc8/goprocam/GoProCamera.py#L124

According to https://github.com/KonradIT/goprowifihack/blob/master/HERO3/WifiCommands.md, it should be a `DF` command instead, and the `%` encoding is only required for passing binary data, it does not make sense to precede a file name with it.

**Proposal:**

Use `DF` instead of `DA`, and either include the `%` in each `sendCamera` call, or do a crude heuristic and only prepend a `%` in `sendCamera` if the `value` argument is a two-digit hexadecimal string.

I can send a PR for the latter.

KonradIT, 2019-01-28:
Good catch, fixed in https://github.com/KonradIT/gopro-py-api/commit/488db6132a8e0989e6e3f10e1c7a42af12ed6f26

closed 2019-01-28 by KonradIT

f0k, 2019-01-28:
That was fast! Did you test it? In https://github.com/KonradIT/goprowifihack/blob/master/HERO3/WifiCommands.md it says something cryptic about using `//` instead of `%OPTION` that I wasn't sure about.

KonradIT, 2019-01-28:
Probably meant to say /XXXGOPRO/FILE.MP4 like on HERO4 cameras and onwards.

f0k, 2019-01-28:
Ok. I'll test it sometime later and report back.