KonradIT/goprowifihack #140: HERO7 Silver

mikinet1, 2019-02-05:
Problem:

I have Hero 7 silver,
and the "http://10.5.5.9:8080/gp/gpMediaList", is not working ,

I get the response:
"{protocol=http/1.1, code=200, message=OK, url=[http://10.5.5.9:8080/gp/gpMediaList}](http://10.5.5.9:8080/gp/gpMediaList%7D) "

I am sending it like this, 
how do i get the media list?
Thanks.

mikinet1, 2019-02-05:
can you send me a sample code that is working, because, i am not sure i am sending the   OkHttpClient correct.

KonradIT, 2019-02-05:
Okay, it seems that the HERO7 Silver differs from the Black in terms of API usage. Do you know how to sniff web traffic? There are multiple ways, but the easiest is to use Packet Capture on Android which works without root and using a VPN. I haven't seen that response yet so I think the API differs.

KonradIT, 2019-02-05:
Try with `curl http://10.5.5.9/gp/gpMediaList` on your PC.

mikinet1, 2019-02-05:
i can't connect the GoPro to my PC. (It's not a laptop.)

KonradIT, 2019-02-05:
Assuming you're doing it on an Android phone, just use Chrome or Firefox to input the URL.

mikinet1, 2019-02-05:
on the browser i get the JSON list.

KonradIT, 2019-02-05:
Neat, the code 200 is for OK so yeah makes sense. Something's wrong with your Android code.

mikinet1, 2019-02-05:
can you help my with that?

KonradIT, 2019-02-05:
Not without more details. In your code, where is the call to /gp/gpMediaList? You're using /videos/DCIM as your base URI but you never call /gp/gpMediaList

KonradIT, 2019-02-05:
Check with my Java API for Android: https://github.com/KonradIT/CamControl/tree/master/mobile/src/main/java/com/chernowii/camcontrol/camera/goproAPI

mikinet1, 2019-02-05:
is it a GET or POST request? 

i try it like this withe the gpmedialist, it's the same,

public class MediaBrowserActivity extends Activity {

    HttpAsyncTask HttpAsyncTask;
    private static Retrofit GoProMainAPI = null;
    private static Retrofit GoProMediaAPI = null;

    @Override
    protected void onCreate(Bundle savedInstanceState)
    {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_media_browser);

        SendOkHttpClient("http://10.5.5.9/gp/gpMediaList");
    }

    public void SendOkHttpClient(String url)
    {
        HttpAsyncTask = new HttpAsyncTask();
        HttpAsyncTask.execute(url);
    }

    private class HttpAsyncTask extends AsyncTask<String, String, String>
    {
        String mystrres;
        @Override
        protected String doInBackground(String... urls)
        {
            return POST(urls);
        }

        @Override
        protected void onPostExecute(String result)
        { }

        public String POST(String... url)
        {
            OkHttpClient client = new OkHttpClient();

            Request request  = new Request.Builder().url(url[0]).build();
            client.newCall(request).enqueue(new Callback() {
                @Override
                public void onFailure(Call call, IOException e)
                {
                    e.printStackTrace();
                }

                @Override
                public void onResponse(Call call, Response response) throws IOException
                {
                    Log.d("LOG",response.message());

                }
            });
            return null;
        }
    }

KonradIT, 2019-02-05:
It's a GET request.

mikinet1, 2019-02-05:
do i need to use this :   Call<GoProMediaList> getMediaList();

KonradIT, 2019-02-05:
Yup.

mikinet1, 2019-02-05:
i get it like this retrofit2.Call<GoProMediaList> getMediaList();

mikinet1, 2019-02-05:
how do i get the response ?

mikinet1, 2019-02-05:
i don't understand how to use it,
      ApiClient GoProApi = ApiBase.getMainClient().create(ApiClient.class);
        retrofit2.Call<GoProMediaList> setting = GoProApi.config(param, value);
        setting.enqueue(new retrofit2.Callback<GoProMediaList>() {
            @Override
            public void onResponse(retrofit2.Call<GoProResponse> call, retrofit2.Response<GoProResponse> response) {
                Log.d("LOG",response.message()); }

            @Override
            public void onFailure(retrofit2.Call<GoProResponse> call, Throwable t) {
            }

        });

mikinet1, 2019-02-05:
what am i doing wrong? 

        ApiClient GoProApi = ApiBase.getMediaClient().create(ApiClient.class);
        retrofit2.Call<GoProMediaList> GetPhotos = GoProApi.getMediaList();
        GetPhotos.execute(new Callback<GoProMediaList> ()
        {

            @Override
            public void onFailure(Call call, IOException e) {
                
            }

            @Override
            public void onResponse(Call call, Response response) throws IOException {

            }
        }

KonradIT, 2019-02-05:
`@Override
        public void onResponse(Call call, Response response) throws IOException {
                // do your stuff here with response.
        }`

closed 2019-02-05 by KonradIT

mikinet1, 2019-02-05:
i still get 
Response{protocol=http/1.1, code=200, message=OK, url=http://10.5.5.9:8080/gp/gpMediaList}