KonradIT/goprowifihack #144: how to get a  specific photo from hero 7

mikinet1, 2019-02-11:
Hi,

is there an API to get a photo to a bitmap in good resolution from the hero 7 silver?

KonradIT, 2019-02-11:
```

gopro.downloadMedia([folder],[file])```

mikinet1, 2019-02-11:
hi,
thanks,
what is gopro? 
don't it have an HTTP GET command for it?
or i did not understand the syntax ?

KonradIT, 2019-02-11:
Ah that is how to do it from my Python wrapper. Check out your camera's Mediabrowsing.md file.

mikinet1, 2019-02-11:
and in android java?

mikinet1, 2019-02-11:
also i can't stream without  a leg in the streaming.
do you have a working sample?

KonradIT, 2019-02-11:
Sorry but this crosses the line of what this repository is for. I'm not here to do your homework. You've created multiple issues which is discouraged. Please do not do that. Also please search before posting.

As per your issues:

- Check here for Android code: https://github.com/KonradIT/CamControl/tree/master/mobile/src/main/java/com/chernowii/camcontrol/camera/goproAPI
and for streaming: https://github.com/KonradIT/gopro-udp-stream-android/
- To get the specific photo: http://10.5.5.9/videos/DCIM/[your folder, eg 100GOPRO]/[your file]

closed 2019-02-11 by KonradIT

mikinet1, 2019-02-11:
Sorry for the mess.....
i am using this :* [http://10.5.5.9/videos/DCIM/[your](http://10.5.5.9/videos/DCIM/100GOPRO]/GOPRO141.JPG)
in my browse, and it's not working , i get error 404: not found.
help please what am i doing wrong ?